schreibfaul1/ESP32-MiniWebRadio #1: Mehr Infos?!

djuscha, 2018-03-07:
Hallo
wird es mehr Infos geben? Was ist das für Display? Hab Edzelf Radio am laufen kann ich es erweitern ? 

Danke

schreibfaul1, 2018-03-07:
Hallo djuscha, ich habe das 2.8inch Display von Waveshire mit eingebauten Adapter für eine MikroSD-Karte benutzt, https://www.waveshare.com/wiki/2.8inch_TFT_Touch_Shield (hab ich bei Amazon gekauft, das MP3 Modul auch), dafür gibt es eine Library in meinem Repository. Prinzipiell geht jedes andere SPI Display mit Touchpad wenn der Code für den Controller angepasst wird. Auf der SD-Karte ist die Senderliste gespeichert, die Senderlogos, die Sprachdateien für die Uhrzeit, die Buttons, mp3 Files u.s.w.  , einfach die zip Datei in additional Info auf die SDCard entpacken. Dort befindet sich eine Dev-Bord.gif mit den benutzten Anschlüssen die verdrahtet werden müssen.  Als IDE am besten Eclipse (Sloeber) verwenden, die Arduino IDE geht sicher auch.  
vG sf1

djuscha, 2018-03-08:
Danke schön  für die Infos!!

closed 2018-03-09 by schreibfaul1