schreibfaul1/ESP32-MiniWebRadio #2: Problem with uploading a sketch

Mateuszz21, 2018-04-14:
Hello, I have a problem with the loading of your sketch I did everything as you described it, I copied everything to one folder but when loading it jumps to me 

Arduino:1.8.5 (Windows 10), Płytka:"NodeMCU-32S, 80MHz, 921600"

C:\Users\HP\Desktop\ESP32-MiniWebRadio-master\MiniWebradio\MiniWebRadio.ino: In function 'void getsettings(int8_t)':

MiniWebRadio:714: error: no matching function for call to 'HTML::reply(String&, bool)'

             web.reply(val, false);

                                 ^

In file included from C:\Users\HP\Desktop\ESP32-MiniWebRadio-master\MiniWebradio\MiniWebRadio.ino:61:0:

sketch\html.h:54:7: note: candidate: void HTML::reply(const String&)

  void reply(const String &response);

       ^

sketch\html.h:54:7: note:   candidate expects 1 argument, 2 provided

MiniWebRadio:720: error: no matching function for call to 'HTML::reply(String&, bool)'

           web.reply(val, false);

                               ^

In file included from C:\Users\HP\Desktop\ESP32-MiniWebRadio-master\MiniWebradio\MiniWebRadio.ino:61:0:

sketch\html.h:54:7: note: candidate: void HTML::reply(const String&)

  void reply(const String &response);

       ^

sketch\html.h:54:7: note:   candidate expects 1 argument, 2 provided

Mateuszz21, 2018-04-14:
I tried and now I have a problem

Arduino:1.8.5 (Windows 10), Płytka:"NodeMCU-32S, 80MHz, 921600"

Archiving built core (caching) in: C:\Users\HP\AppData\Local\Temp\arduino_cache_333205\core\core_espressif_esp32_nodemcu-32s_FlashFreq_80,UploadSpeed_921600_508f2172c8b27b00ad4af7a35a8fefc9.a
sketch\IR.cpp.cpp.o:(.iram1.literal+0x54): undefined reference to `ir'

collect2.exe: error: ld returned 1 exit status

schreibfaul1, 2018-04-15:
Hello Mateuszz21,
on my last commit  I forgot upload the modified html.cpp and html.h. This is now done. Sorry for that.
I want in the future the websurface rebuild, for that is now necessairy that not every web,reply() send the html-header.

closed 2018-04-15 by schreibfaul1