schreibfaul1/ESP32-MiniWebRadio #3: Firmware file

folkos, 2018-04-28:
Hi . You can upload the firmware file in the bin format.

schreibfaul1, 2018-04-29:
https://drive.google.com/open?id=1qI1V-FKkn_AY_FtgGtS1lVXbFSUJ_vJH

schreibfaul1, 2018-04-29:
Hello folkos,
when using the binary files, the hardware must be exactly mine. You cannot change anything, do not switch GPIOs, do not use a display with another controller. That may be good for a first test. The better way is to use a programming environment, such as Eclipse or the Arduino IDE. A video on installing Eclipse / Sloeber You can find in YouTube. For the Arduino-IDE is a help in the additional infos. Test each component individually, the MP3 module, the display....
Only when each component works properly pack everything together. This makes it easier to find errors and errors always happen, especially in a larger projects. And remember: the default. csv needs to be adjusted, the web radio needs more memory.
Regards
sf1

closed 2018-04-29 by schreibfaul1

Mateuszz21, 2018-05-01:
Hello, I have all the parts like you and I do everything according to your guide and instructions, the program goes well, but I do not read the SD card and see only "I can not connect to Wi-Fi, check your credentials" on the display
what's going on, why?