schreibfaul1/ESP32-MiniWebRadio #4: Problem with WIFI connectivity

Mateuszz21, 2018-05-01:
Hello, I have all the parts like you and I do everything according to your guide and instructions, the program goes well, but I do not read the SD card and see only "I can not connect to Wi-Fi, check your credentials" on the display
what's going on, why?

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
tft_info: TFT_CS:22 TFT_DC:21 TFT_BL:17 TFT_MOSI:23 TFT_MISO:19 TFT_SCK:18
tft_info: init HX8347D

schreibfaul1, 2018-05-03:
Is that all, mo more lines?

Mateuszz21, 2018-05-03:
The problem is that he does not write anything more.
How to install the code "https://drive.google.com/open?id=1qI1V-FKkn_AY_FtgGtS1lVXbFSUJ_vJH", it finds the network, but it does not connect to the right one, even though I changed it everywhere

schreibfaul1, 2018-05-03:
I have tested he binary files with ESP32 boards form Widora, Espressif and Doit.  Never the error can be produced. The new PicoD4 Board will not work beacause pin17 is  internally connected.  Please run the scourcecode in Your IDE ans see wat happend. So You can find out what is wrong.

Noelle88, 2018-05-18:
Guten Abend.

Heute kamen auch meine letzten Komponenten an.
Leider musste ich feststellen, dass bei mir genau der selbe Fehler auftritt wie bei Mateuszz21.

DevBoard ist dieses hier:
https://www.ebay.de/itm/Espressif-ESP32-WLAN-Dev-Kit-Board-Development-Bluetooth-Wifi-v1-WROOM32-NodeMCU/253564809807?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

- Hardware 3-4 mal auf Fehler geprüft -> alles i.o.
- statt Spannungsregler bekommt die Steuerung jetzt vom Labornetzteil seine 5V
- SD-Card mehrmals Formatiert

Serial Monitor spuckt bei mir aber noch einen Hinweis mehr aus:

ets Jun  8 2016 00:22:57

**rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57**

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
tft_info: TFT_CS:22 TFT_DC:21 TFT_BL:17 TFT_MOSI:23 TFT_MISO:19 TFT_SCK:18
tft_info: init HX8347D

Und noch etwas ist mir aufgefallen:

Ich habe spaßeshalber das VS1053 getrennt., da kam nur noch Ameisenfußball auf dem LCD.
Nach dem er wieder angeschlossen war, kam wieder "can´t connect to WiFi, check Your credentials"

Ich bin behüht den fehler zu finden, bin aber auch nicht so fitt in der Programmierung und es ist mein erstes Projekt mit dem ESP32.

Vielleicht hilft meine Beschreibung bei der Fehlersuche.
Falls ich weiter komme, werde ich es auf jeden fall berichten.

Noelle88, 2018-05-20:
Ich habe heute noch mal ein wenig rum probiert....

Ich habe aus Ihrem älteren Youtube-Video zum "MiniWebRadio mit ESP32" den Code abgeschrieben und siehe da... hat auch auf anhieb funktioniert.
Das TFT habe ich nicht noch mal extra getestet, weil das ja bis zum Startbildschirm und der Fehlermeldung funktioniert.

Sloeber habe ich auch installiert. Wenn ich damit das Programm überprüfe, zeigt er mir einen Fehler in Zeile 692  (Connect To WiFi ) "WiFiName=WiFi.SSID(i);" an.
Als Warnung wird folgendes ausgegeben:
_Multiple markers at this Line - Write occurrence of `SSID` - Invalid arguments 'Candidates are: String SSID()'_

Eigentlich hatte ich vor meine Zugangsdaten einfach mal direkt im Code zu plazieren, hat aber eben auf die Schnelle leider nicht geklappt.

schreibfaul1, 2018-05-26:
Die Verbindung zum WLAN wird nun über wifimulti.h hergestellt.  Die Funktionalität ist ähnlich. Es wird der stärkste AP gesucht, zu dem die Zugangsdaten bekannt sind.

closed 2018-05-26 by schreibfaul1