schreibfaul1/ESP32-MiniWebRadio #5: Touch calibration

gonza88, 2018-05-13:
Hello
I made this radio and I have problem with cheap LCD based on ILI9341. When I touch the screen, point of touch doesn't match to icons of menu etc. It's possible to calibrate this?

PS. It's looks like ESP don't read presets.csv file from SD card. Radio plays only stations preprogrammed in ino file.

schreibfaul1, 2018-05-14:
Hello,
You use my tft library? If so scroll tft.cpp to the end and seek for 
x = TP_Send(0xD0);
y=  TP_Send(0x90); 
read x,y with (0,0) and (320,240) then You have th Voltage (in milliVolts) from TP controller.
Replace the constants Xmin, Xmax, Ymin, Ymax in tft.h with Your values.

In Your browser, tab config try the button "load default" to reread presets.csv.

schreibfaul1, 2018-05-19:
maybe You have this display: TJCTM24028-SPI, ILI9341, red board?
tochpad is y-inverted. If so replace the code at the end tft.cpp
  if(_rotation==0){y=TFT_HEIGHT-y;}
  if(_rotation==1){tmpxy=x; x=y; y=tmpxy; y=TFT_WIDTH-y; x=TFT_HEIGHT-x;}
  if(_rotation==2){x=TFT_WIDTH-x;}
  if(_rotation==3){;}  // do nothing

and set in tft.h:
      const uint16_t Xmax=1850;
      const uint16_t Xmin=170;
      const uint16_t Ymax=1880;
      const uint16_t Ymin=140;

koka7777745, 2018-07-15:
Dear author, tell me please how to add a red board for ILI9341, support for Russian-language fonts. And then I have at Russian stations some kryakozyabry.

koka7777745, 2018-07-15:
Thanks for the tips, 11 commented, and 8 commented. The rest does not come. Compilation does not go through the same mistakes. Drop the corrected files to the mail.
koka5675766@rambler.ru

koka7777745, 2018-07-15:
And how are the logos attached to the stations?

schreibfaul1, 2018-08-12:
To translate all the titles or words in russian You can try this:

  tft.setFont(Garamond_cyrillic_32x33);
  tft.print(UTF8toCp1251("Калинка, калинка, калинка моя! В саду ягода малинка, малинка моя!"));
  tft.print(UTF8toCp1251("Ах, под сосною, под зелёною, Спать положите вы меня!")); 

for example: 

in setup()
``
// showHeadlineItem("** Internet radio **");
showHeadlineItem("ИНТЕРНЕТ-РАДИО");
``

and in showHeadlineItem(const char* hl)
``
// tft.setTextSize(2);
tft.setFont(Garamond_cyrillic_18x18);
display_info(UTF8toCp1251(hl), _yHeader, _hHeader, TFT_WHITE, 0);
...
``

pilnikov, 2018-08-13:
' void showTitle(String str) {'
'  static String title = "";'
'  str.trim();  // remove all leading or trailing whitespaces'
'  if ((_state == RADIO) && (title == str)) return; // nothing to do'
'  if (str.length() > 4) f_has_ST = true; else f_has_ST = false;'
'  /* tft.setTextSize(4);'
'     if(str.length()> 45) tft.setTextSize(3);'
'     if(str.length()> 80) tft.setTextSize(2);'
'     if(str.length()>100) tft.setTextSize(1);'
'  */'
''
'  tft.setFont(Garamond41x42cyrillic);'
'  if (str.length() > 45) tft.setFont(Garamond32x33cyrillic);'
'  if (str.length() > 80) tft.setFont(Garamond23x24cyrillic);'
'  if (str.length() > 100) tft.setFont(Garamond21x21cyrillic);'
''
'  displayinfo(str.c_str(), _yTitle, _hTitle, TFT_CYAN, 0);'
'  title = str;'
'}'
' void showStation() {'
 ' tft.setFont(Garamond41x42cyrillic);'
 ' if (_stationname.length() > 45) tft.setFont(Garamond32x33cyrillic);'
 ' if (_stationname.length() > 80) tft.setFont(Garamond23x24cyrillic);'
 ' if (_stationname.length() > 100) tft.setFont(Garamond21x21cyrillic);'
''
 ' String str1 = "", str2 = "";'
 ' if (_stationname == "") {'
 '   //    tft.setTextSize(3);'
 '   //    if (_station.length() > 75) tft.setTextSize(1);'
 '   displayinfo(_station.c_str(), _yName, _hName, TFT_YELLOW, _wLogo + 14); // Show station name'
 '   showTitle("");   // and delete showstreamtitle'
 '   showFooter();'
 '   str1 = _station;'
 ' } else {'
 '   //   tft.setTextSize(4);'
 '   //   if (_stationname.length() > 30) tft.setTextSize(3);'
 '   displayinfo(_stationname.c_str(), _yName, _hName, TFT_YELLOW, _wLogo + 14);'
 '   showTitle("");'
 '   showFooter();'
 '   str1 = _stationname;'
 ' }'
 ' //log_i("%s", _stationname.c_str());'
 ' str1.toLowerCase();'
 ' str1.replace(",", ".");'
 ' str2 = "/logo/" + String(UTF8toASCII(str1.c_str())) + ".bmp";'
 ' //log_i("%s", _sbuf);'
 ' if (f_SD_okay) if (tft.drawBmpFile(SD, str2.c_str(), 0, _yLogo) == false) tft.drawBmpFile(SD, ' ' "/logo/unknown.bmp", 1, 22);'
'}`
Right name of cyrillic font is Garamond32x33cyrillic

schreibfaul1, 2018-08-20:
Yes, use fontname Garamond32x33cyrillic

closed 2018-08-20 by schreibfaul1