schreibfaul1/ESP32-MiniWebRadio #6: LOGO

Mateuszz21, 2018-05-27:
Hello, everything works for me just how to add a logo to the SD card at what resolution, because, as I added this radio does not display the proper icon only logo "/logo/unknown.bmp" ??

schreibfaul1, 2018-05-27:
Hello Mateuszz21,
all logos have the same resolution 96x96 px, color 24 bit. I have generated the logos with paint.net. The filename must be the stationname in presets.csv. New stations, not in presets: the filename is that, what You can see on the display. Replace commas with dots. All ASCII characters öüäßèñë.. can be used in filenames.

schreibfaul1, 2018-06-14:
My tft library handle also  jpg. You can convert all *.bmp into *.jpg (e.g. with ImageBatch), it works faster.

closed 2018-06-14 by schreibfaul1