schreibfaul1/ESP32-MiniWebRadio #7: Can not compile with Arduino 1.8.5 under windows

franki29, 2018-06-12:
Hi, i tried to compile the code but always get the same error:

Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Default, QIO, 80MHz, 4MB (32Mb), 921600, None"

MiniWebRadio:134: error: expected primary-expression before 'void'

 void defaultsettings(){

 ^

MiniWebRadio:134: error: expected '}' before 'void'

MiniWebRadio:134: error: expected ',' or ';' before 'void'

MiniWebRadio:134: error: expected unqualified-id before numeric constant

 void defaultsettings(){

    ^

MiniWebRadio:475: error: expected declaration before '}' token

     tft.print(rtc.gettime_s());

 ^

Mehrere Bibliotheken wurden für "SD.h" gefunden
 Benutzt: C:\Users\frank\Documents\Arduino\hardware\espressif\esp32\libraries\SD
 Nicht benutzt: C:\Users\frank\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\SD
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\frank\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Nicht benutzt: C:\Users\frank\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
exit status 1
expected primary-expression before 'void'

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre

thanks for helping
Frank

schreibfaul1, 2018-06-14:
Hallo Frank,
es könnte helfen, das gesamte Projekt neu zu kompilieren. Da die Arduino IDE kein build all, clear, rebuild index... kennt kannst du einfach ein ähnliches Board auswählen und den Kompiliervorgang neu starten.

closed 2018-06-14 by schreibfaul1

franki29, 2018-06-14:
Hallo Wolle,
geht leider immer noch nicht :

Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Default, DOUT, 80MHz, 4MB (32Mb), 921600, None"

Build-Optionen wurden verändert, alles wird neu kompiliert
MiniWebRadio:142: error: expected primary-expression before 'void'

 0void defaultsettings(){

 ^

MiniWebRadio:142: error: expected '}' before 'void'

MiniWebRadio:142: error: expected ',' or ';' before 'void'

MiniWebRadio:142: error: expected unqualified-id before numeric constant

 0void defaultsettings(){

    ^

MiniWebRadio:475: error: expected declaration before '}' token

         displayinfo(_stationname.c_str(), _yName, _hName, TFT_YELLOW, _wLogo+14);

 ^

Mehrere Bibliotheken wurden für "SD.h" gefunden
 Benutzt: C:\Users\frank\Documents\Arduino\hardware\espressif\esp32\libraries\SD
 Nicht benutzt: C:\Users\frank\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\SD
Mehrere Bibliotheken wurden für "WiFiClient.h" gefunden
 Benutzt: C:\Users\frank\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Nicht benutzt: C:\Users\frank\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
 Nicht benutzt: C:\Users\frank\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
 Nicht benutzt: C:\Users\frank\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
 Nicht benutzt: C:\Users\frank\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
exit status 1
expected primary-expression before 'void'

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.