schreibfaul1/ESP32-MiniWebRadio #9: Programmspeicherplatz reicht nicht aus?

Noelle88, 2018-06-22:
Guten Abend!  :-)

Nach dem mein Radio nun 3 Wochen ohne Probleme lief (Danke noch mal für die Hilfe),
sah ich dass es neue Dateien gibt, die ich eben Aufspielen wollte.
Leider meckert die Arduino IDE, dass der Programmspeicher nicht ausreichen würde?! :/
(auch TFT, IR, vs1053_ext wurden gegen die aktuellen getauscht)

_"Der Sketch verwendet 1324138 Bytes (101%) des Programmspeicherplatzes.
Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 69240 Bytes (23%) des dynamischen Speichers,
225672 Bytes für lokale Variablen verbleiben. Das Maximum sind 294912 Bytes."_

Kommt das denn hin mit den 1324138 Bytes?  Wenn ja, wurde ich wohl beim Board beschissen :D

schreibfaul1, 2018-06-23:
Hallo Noelle88,
der ESP32 bingt in der Regel einen 520 kBit SRAM (Arbeitsspeicher) und 32 MBit Flash (Programmspeicher)= 4Mbyte mit. 
Durch das Vergrößern des Partition (factory app) in der default.cvs auf 3Mbyte ist ausreichend Platz vorhanden. 
Die Arduino-IDE setzt eine willkürliche Grenze. Diese steht in der boards.txt (arduino/hardware/espressif/esp32). Dort steht z.B. für das ESP32 Dev Module und für alle anderen Boards:
esp32.upload.maximum_size=1310720.

Ändere bitte den Wert auf 3145728

Noelle88, 2018-06-23:
Super! Herzlichen Dank, das war der Fehler :-)

closed 2018-07-09 by schreibfaul1