schreibfaul1/ESP32-MiniWebRadio #12: Kein Touch

Feuer001, 2018-07-27:
Hey muss nochmal stören,

meine ILI9341 Display mit Touch sind gekommen, aber der Touch zeigt keine Reaktion.
Muss ich da noch was beachten? Habe in der .ino Datei auf tot(0) gestellt.
Ist doch richtig das von dem Touch nur zwei Pin benötigt werde, oder?
Also nur TP_IRQ und TP_CS??

schreibfaul1, 2018-07-27:
ja, das ist richtig, T_IRQ (GPIO39) und T_CS (GPIO16)
und dann noch T_DO mit SPI_MISO (GPIO19), T_DI mit SPI_MOSI (GPIO23) und T_CLK mit SPI_CLK (GPIO18)

Feuer001, 2018-07-27:
ja ok, dann kann es ja auch nicht funktionieren, hab die T_DO/DI/CLK vom touch nicht angeschlossen...
Wenn ich die dann anschließe wird es wohl laufen, kann ich im Moment aber nicht testen, Lötkolben sind schon für den Umzug eingepackt. Muss ich mich wohl noch etwas gedulden.
Besten dank an dich für deine Super Unterstützung !!

Feuer001, 2018-07-30:
Also der Touch reagiert, aber irgendwie nicht richtig...
Wenn mir die Symbole angezeigt werden und ich auf das Stumm-Symbol drücke kommt die nächste Symbolseite. Gibt es bei den Touch-Display verschiedene Touch-Treiber??

schreibfaul1, 2018-07-30:
Es wird der 2046  Touchcontroller verbaut sein. Es kann sein dass x,y vertauscht ist. Oben links sollte 0,0 sein. In  der Library gibt es ein kleines Testprogramm.

```
#include "Arduino.h"
#include "SPI.h"
#include "tft.h"

#define TP_IRQ        39
#define TP_CS         16

TFT tft(0); // (0) ILI9341 Display, (1) Waveshare 2.8 TFT with TP
TP tp(TP_CS, TP_IRQ);

uint16_t tp_x, tp_y;

void setup() {
    SPI.begin();
    tft.begin();
    tft.setRotation(3); // Use landscape format
    tp.setRotation(3);
    tft.fillScreen(TFT_BLACK);
    tft.setTextColor(TFT_GREENYELLOW);
    tft.setTextSize(2);
}

//-------------------------------------------------------------------------------------
void loop(void) {
    tp.loop();
}
//-------------------------------------------------------------------------------------

// Event from TouchPad
void tp_pressed(uint16_t x, uint16_t y){
    tp_x=x;  tp_y=y;
}
void tp_released(){
    tft.fillRect(100, 100, 80, 40, TFT_BLACK);
    tft.setCursor(100, 100);
    tft.print("PosX="); tft.println(tp_x);
    tft.print("PosY="); tft.println(tp_y);
}
```

Dabei muss das VS1053 Board gezogen werden oder VS1053_CS muss ständig HIGH sein

Feuer001, 2018-07-30:
hey, das Display läuft, habe in der tft.ccp die rotations Reihenfolge für den touch geändert.
Normal ist ja 0,1,2,3 und nun 2,3,0,1. (habe die stelle wo die Rotation in der main angegeben ist nicht gefunden...) 
Alles super !!
Aber noch eine Frage hätte ich, wäre es vielleicht möglich so eine art Standby modus einzubauen, in dem auf dem Display die Uhrzeit sowie im Wecker angezeigt wird und gleichzeitig, der VS1053 und der Verstärker abgeschaltet wird?

schreibfaul1, 2018-07-30:
in setup()  tp.setRotation(3); wird dann bei deinem Display tp.setRotation(1); sein
Somit gibt es zwei unterschiedliche Versionen von dem Display, eine Version mit um 180 Grad gedrehtem Touchpad, meins ist von Amazon.
Bei Waveshare Display gibt es einen Transistor für die Steuerung  Hintergrundbeleuchtung, hier nicht. BL muss für eine ordentliche Helligkeit auf +5V gezogen werden, der ESP schafft max 3,3V. Ich werde einen Inverter vorschalten oder einen OPV damit die Dimmfunktion richtig funktioniert.
Standby kommt dann in einer späteren Version.

Gruß Wolle

schreibfaul1, 2018-08-12:
Neu gibt es eine Einschlafautomatik, nach einer in Intervallen einstellbaren Zeit schaltet das Radio ab und wacht erst wieder auf nachdem das Display berührt wurde.

closed 2018-08-12 by schreibfaul1