schreibfaul1/ESP32-MiniWebRadio #14: Touch screen does not work

gersey, 2018-08-19:
Hello,
This is a very good project. Thanks for it.
But my touch screen does not work. I also tried your little test program, but it does not work either.
I use this screen (HR2046): https://bit.ly/2BqfedC
Wire: T_IRQ (GPIO39), T_CS (GPIO16), T_DO (GPIO19), T_DI (GPIO23), T_CLK (GPIO18).

What could be the problem?
Thanx

schreibfaul1, 2018-08-20:
The connections are correct. I think the TFT display should work. If all components are connected, the tesprogramm only  works if all other CS-pins  are high, so add 

#define VS1053_CS 2
#define SD_CS 5

and in setup()

    pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);
    pinMode(VS1053_CS, OUTPUT);  digitalWrite(VS1053_CS, HIGH);

gersey, 2018-08-20:
I tried a little:
The test program runs perfectly with VS1053 (#define VS1053_CS 2 and pinMode(VS1053_CS, OUTPUT);  digitalWrite(VS1053_CS, HIGH);),
but if I connect microsd card adapter it will be wrong (Even if it includes: #define SD_CS 5 and pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);)

schreibfaul1, 2018-08-20:
Sd Cards are sometimes difficult, MISO and MOSI needs a pullup resistor. On my display are pullups already integrated. Perhaps these not included in all displays.
[image]

gersey, 2018-08-21:
I use catalex microsd card adapter.

gersey, 2018-08-21:
I now read that it needs to be modified. I did it, it works, but read the pictures very slowly.

schreibfaul1, 2018-08-21:
Without modification calalex does not work together with other components on the same SPI. All wires must be short as possible.

closed 2018-08-24 by schreibfaul1