schreibfaul1/ESP32-MiniWebRadio #15: request for PCM5102 & links in .m3u8 format

vu2iia, 2018-08-26:
Hi,

Will you add support for DAC PCM5102?.  

Will webradio station in .m3u8 format work? 
I am trying this station-   http://airfmgold-lh.akamaihd.net/i/fmgold_1@507591/master.m3u8

This is a wonderful project.
Thanks in  advance.
max

schreibfaul1, 2018-08-26:
Hello,
http://airfmgold-lh.akamaihd.net/i/fmgold_1@507591/master.m3u8 is a mp4 transportstream, it can not work with miniwebradio. Use DVBViewer, VLC media player instead.
PCM5102 needs I2S, a software mp3 decoder and DAC.  I think ESP32 8-bit DAC allows no good sound quality.

iskanderpol, 2018-09-13:
Hello, tell me where and what icons for the interface to load, I realized that the display of icons are stored on the SD card?

schreibfaul1, 2018-09-13:
All graphics are stored on SD, all radiologos, buttons, digits for the alarm clock, some pictures and png for the webpage.

schreibfaul1, 2018-10-02:
in my new version You can seek a suitable stationlogo (e.g. google/image) ,copy the URL and the new image will be fittet to a 96px x 96px jpeg, then give it a proper name and upload it directly to SD

closed 2018-10-02 by schreibfaul1