schreibfaul1/ESP32-MiniWebRadio #16: Help Me

fourchtein, 2018-09-30:
First Tanks for your woks it is very nice 
can you help me ?
compilation ok no pb , transfert no error 
run : rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Fatal exception (29): StoreProhibited
epc1=0x4008030e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000008e, depc=0x00000000
 test on wemos card 32
and 32 dev card 30 pin same than you.

test with .bin ok but no WIFI i put on .cvs SSID|mypassword but program runCan you help me Tank you
Philippe

schreibfaul1, 2018-09-30:
Have You change the default.csv? can be it is not enough space in the factory or nvs partition

fourchtein, 2018-10-01:
TANK S
It Work perfectly very very Nice job 
Thank you ...........

Philippe Fourchtein

closed 2018-10-01 by schreibfaul1