schreibfaul1/ESP32-MiniWebRadio #17: Complementary development proposal,

fourchtein, 2018-10-09:
sincere compliment 
great development 
you have helped me in a valuable way
all my thanks

I have a complementary development proposal,

audio menu choose classic output or Bluetooth to send over Bluetooth headphone

do you think that's feasible?

Thank you again
Philippe from Paris

schreibfaul1, 2018-10-09:
Hello fourchtein,
connect to a Bluetooth audio device would be a nice feature. But it is very difficult and hard to make it. I need a SW mp3 decoder for this purpose (VS1053 can do it also, pin 25/36 I2S MCLK/ I2S R-L out, but the pins are not connected on my board) Look for ESP32 SW MP3 decoder, there is a project on Github, I have never tested it, may be it is useful.

schreibfaul1, 2018-12-22:
Two SW-MP3 decoders were tested, the HELIX and the MAD decoder. Both did not produce completely satisfactory results. The HELIX decoder has been packaged in a class and can be tested, https://github.com/schreibfaul1/ESP32-audioI2S . An external I2S DAC is required. It's enough for a kitchen radio. Therefore, I think (Bluetooth) Bluetooth is not realized for the time being.

closed 2018-12-22 by schreibfaul1