schreibfaul1/ESP32-MiniWebRadio #18: ILI9486 3.5 display support

stpavel, 2018-10-14:
Can you include ILI9486 3.5 display support for miniwebradio ?

schreibfaul1, 2018-10-15:
My TFT-Library-ILI9486 will work with miniwebradio. You need a external SD card adapter, catalex must be modified, see https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/additional_info/CATALEX%20MicroSD%20Card%20Adapter.GIF
Greater logos and fonts can be used. Perhaps I will include this display later.

schreibfaul1, 2018-10-21:
My TFT library ILI9486 works with miniwebradio, 
set
//TFT tft(1);                    
TFT tft;

in setup()
//tft.begin(TFT_CS, TFT_DC, SPI_MOSI, SPI_MISO, SPI_SCK, TFT_BL);
tft.begin(TFT_CS, TFT_DC, SPI_MOSI, SPI_MISO, SPI_SCK);    // Init TFT interface
because the display has no backlight pin

[image]

and now adapt the fonts/graphics to the greater resolution

closed 2018-10-21 by schreibfaul1

dd7lp, 2018-11-18:
Guten Tag, ich suche für das Miniwebradio, welches ich gerne nachbauen möchte die Datei
fonts.h, die im Arduino sketch stehen soll, ich finde die leider nirgend wo. Vielleicht können wir ja mal in kontakt treten. wäre schön wenn sie antworten würden
mfg petersen

Absender dieser E-mail
DD7LP
Christian Petersen
Lassensstieg 4
25853 Bohmstedt
Tel. 04671-5101  mob. 01701137601
Mail:dd7lp1@gmail.com