schreibfaul1/ESP32-MiniWebRadio #20: Mini Web-Radio mit esp32 und Ili 9341

dd7lp, 2018-12-02:
Hallo und guten Abend, ich hatte mich letzte Woche schon mal gemeldet, ich würde gerne dein Projekt mini-web-radio nachbauen, bekomme aber die Software nicht kompiliert. Ich habe heute gesehen, das eine anderer "Nachbauer" die gleichen Probleme hat. Ich wollte ein Diplay 320x240 mit ili9341 chip und Touchscreen verwenden. Ich habe die neuste Version deiner Software von Gitthub hier geladen und bekomme beim kompilieren eine Menge Fehler, wenn ich das Display verwenden will.
hier malo die Fehlermeldungen aus dem Compiler (Arduino 1,85), die Librarys für den esp32 liegen richtig, ich habe schon einiges am laufen. 
Vielleicht hast du einen Tipp für mich, das wäre sehr freundlich. du kannst mich auch direkt anschreiben unter dd7lp1 at gmail.com.
hier die Fehler zu deinem unverändertem Original-Listing 
MiniWebRadio:166: error: no matching function for call to 'TFT::TFT(int)'

 TFT tft(0);                     // parameter:  (0)ILI9341, (1)HX8347D

          ^

In file included from C:\Users\dd7lp\Documents\Arduino\MiniWebRadio\MiniWebRadio.ino:74:0:

sketch\tft.h:88:9: note: candidate: TFT::TFT()

         TFT();

         ^

sketch\tft.h:88:9: note:   candidate expects 0 arguments, 1 provided

sketch\tft.h:85:7: note: candidate: constexpr TFT::TFT(const TFT&)

 class TFT : public Print {

       ^

sketch\tft.h:85:7: note:   no known conversion for argument 1 from 'int' to 'const TFT&'

C:\Users\dd7lp\Documents\Arduino\MiniWebRadio\MiniWebRadio.ino: In function 'void showTitle(String)':

MiniWebRadio:391: error: 'Times_New_Roman43x35' was not declared in this scope

     tft.setFont(Times_New_Roman43x35);

                 ^

MiniWebRadio:392: error: 'Times_New_Roman38x31' was not declared in this scope

     if(txtlen(str)> 30) tft.setFont(Times_New_Roman38x31);

                                     ^

MiniWebRadio:393: error: 'Times_New_Roman34x27' was not declared in this scope

     if(txtlen(str)> 45) tft.setFont(Times_New_Roman34x27);

                                     ^

MiniWebRadio:394: error: 'Times_New_Roman27x21' was not declared in this scope

     if(txtlen(str)> 65) tft.setFont(Times_New_Roman27x21);

                                     ^

MiniWebRadio:395: error: 'Times_New_Roman21x17' was not declared in this scope

     if(txtlen(str)>130) tft.setFont(Times_New_Roman21x17);

                             ******************************************************************
ein wenig Hife wäre toll,
Gruß Christian aus der Nähe von Husum

schreibfaul1, 2018-12-08:
Hallo Christian,
in tft.h Zeile 23 muss der Kommentar entfernt werden
//#include "fonts/Times_New_Roman.h"        // latin, greek, cyrillic with all extensions
Dieser Zeichensatz benötigt viel Flash Speicher und funktioniert nur wenn die Partitionstabelle angepasst wird, mit der default.cvs wird es eng. 
In MiniWebRadio gibt es die geänderte Partitionstabelle.

schreibfaul1, 2018-12-22:
In einem japanischen Blog läuft ein Display mit ILI9341. http://www.love-hokuto.com/?p=7255

closed 2018-12-22 by schreibfaul1

dd7lp, 2018-12-22:
hallo danke schön für deine Mitteilung ich habe das WLAN Radio mittlerweile
am Laufen ich betreibe es mit einem in e93 41 Display was zurzeit noch
nicht richtig will ist die Steuerung mit der Fernbedienung die
fernbedienungscodes werden nicht richtig erkannt aber da muss ich mal
suchen danke schön nochmals für deine Hilfe ein frohes weihnachtsfest und
einen guten Rutsch ins neue Jahr Grüße aus Nordfriesland Christian Petersen