schreibfaul1/ESP32-MiniWebRadio #21: PNG images broken in All Browser Types

GerryGit, 2018-12-17:
Thanks for a great and very complete project. I have most elements working well. 
My problem is that all png buttons are showing as broken placeholders. It is possible to see the partially load before breaking. The behavior is the same on Chrome, Firefox, Safari. I have tried with antivirus and firewalls off. By right clicking and opening image in a new tab I can see an icon which is only partially downloaded. Image attached. It feels like an html  timing issue, in that the image does not fully download. All of the images are fully visible on the SD card.  I am using a LOLIN ESP32 board. 
Any help would be appreciated. 
[image]
[image]

schreibfaul1, 2018-12-17:
This is a bug that occurs with the older WiFi libraries. Please use the version from July 2018
https://github.com/espressif/arduino-esp32

GerryGit, 2018-12-18:
Thanks Wolle. I did try that and have installed esp32 1.0.0 .
I also updated the preferences.txt in the IDE.
Unfortunately this does not seem to have solved the problem.
It is still exactly the same. I am using Arduino IDE 1.8.5

Any further suggestions?

Thanks.

GerryGit, 2018-12-18:
Also tried 1.0.0-rc2 and -rc4
Behavior is the same.

schreibfaul1, 2018-12-18:
Hello Gerry, I've recompiled the project in the Arduino IDE with all libraries from Github and can not reproduce the error. With the old WiFi libraries, the transmission stopped before the file was completely sent. Therefore, I know this error
[image]

schreibfaul1, 2018-12-18:
perhaps I found a solution for missing ACK, see https://github.com/esp8266/Arduino/issues/1853, 
html.cpp is updated,
(line 91 in sub streamfile()  )

GerryGit, 2018-12-20:
Wolle. I have solved the Arduino library spaghetti by moving to my clean laptop. Problem solved! Once again thanks for a great and feature-full project. 
[image]

closed 2018-12-20 by GerryGit