schreibfaul1/ESP32-MiniWebRadio #22: miniwebradio sd Karte wird nicht gefunden.

urmel33, 2018-12-23:
Hallo, habe mein ESP32 Dev. Board und das waveshare 2.8 nach der Anleitung verbunden und die Dateien auf eine SD gespielt.
Leider läuft das Ganze bei mir nicht. Laut Seriellen Monitor wir die SD Karte nicht gefunden:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
setup      : Init SD card
tft_info   : TFT_CS:22 TFT_DC:21 TFT_BL:17 TFT_MOSI:23 TFT_MISO:19 TFT_SCK:18
tft_info   : init HX8347D
setup       : SD card not found

SD Karte steckt im Kartenslot des waveshare 2.8.

Wäre nett wenn sie mir weiterhelfen könnten. DANKE

urmel33, 2018-12-30:
Sorry, das ich mich so spät melde. Danke für den Tipp, werde das morgen noch mal überprüfen. 
Melde mich dann.

hbulf, 2019-08-26:
Hallo,
erst mal vielen dank für das super Projekt, allerdings bekomme ich nach dem kompilieren und reset diesen Fehler.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Fatal exception (29): StoreProhibited
epc1=0x4008033c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000008fb, depc=0x00000000
Können sie mir da weiterhelfen?

closed 2019-12-31 by stale[bot]