schreibfaul1/ESP32-MiniWebRadio #25: not an issue! Only for Windows user? pitty

Robiv8, 2019-04-23:


schreibfaul1, 2019-05-01:
You can install Sloeber or Arduino-IDE in many OS
hera a example to install sloeber on Ububtu (Linux Mint):
http://eclipse.baeyens.it/stable.php?OS=Linux
[image]
download and extract, start sloeber-ide
[image]
define a workspace folder and launch
[image]
goto Window – Preferences – Third party index url‘s and add (oher entries can be deleted)
https://github.com/espressif/arduino-esp32/releases/download/1.0.2/package_esp32_dev_index.json
[image]
then Platforms and Boards, choose a Version:
[image]
Add the package Python-serial
[image]
and give access to the USB-Port  dmesg | grep tty
[image]

and now create a ESP32 project ...

closed 2019-05-01 by schreibfaul1

Robiv8, 2019-05-03:
Thanks for explanation