schreibfaul1/ESP32-MiniWebRadio #26: SPI TFT change

alansbor, 2019-05-17:
Hello

I tried to change the screen connection from your option

// Digital I/O used
#define VS1053_CS      2  // do not use GPI032 or GPIO33 here
#define VS1053_DCS     4  // do not use GPI032 or GPIO33 here
#define VS1053_DREQ   36
#define TFT_CS        22  // do not use GPI032 or GPIO33 here
#define TFT_DC        21  // do not use GPI032 or GPIO33 here
#define TFT_BL        17  // 33 (pico V4)
#define TP_IRQ        39
#define TP_CS         16  // 32 (pico V4)
#define SD_CS          5
#define IR_PIN        34
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18

To their values:

#define VS1053_CS      2  // do not use GPI032 or GPIO33 here
#define VS1053_DCS     4  // do not use GPI032 or GPIO33 here
#define VS1053_DREQ   36
#define TFT_CS        15 //22  // do not use GPI032 or GPIO33 here
#define TFT_DC        16 //21  // do not use GPI032 or GPIO33 here
#define TFT_BL        17  // 33 (pico V4)
#define TP_IRQ        39
#define TP_CS         16  // 32 (pico V4)
#define SD_CS          5
#define IR_PIN        34
#define SPI_MOSI      13 //23
#define SPI_MISO      12 //19
#define SPI_SCK       14 //18

The program simply does not want to work with them. Why? Can I fix it?

schreibfaul1, 2019-05-20:
Hello,
if you use ports other than the default ports for SPI, speed reduction can help. See https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html

[image]

alansbor, 2019-05-20:
Hello,

Your call takes place in the tft.cpp file

`270 SPI.begin (TFT_SCK, TFT_MISO, TFT_MOSI, -1);`

And in the file SPI.CPP

```
37 void SPIClass :: begin (int8_t sck, int8_t miso, int8_t mosi, int8_t ss)
38 {
    if (_spi) {
        return;
    }

    if (! _ div) {
        _div = spiFrequencyToClockDiv (_freq);
    }

    _spi = spiStartBus (_spi_num, _div, SPI_MODE0, SPI_MSBFIRST);
    if (! _ spi) {
        return;
    }

    if (sck == -1 && miso == -1 && mosi == -1 && ss == -1) {
        _sck = (_spi_num == VSPI)? SCK: 14;
        _miso = (_spi_num == VSPI)? MISO: 12;
        _mosi = (_spi_num == VSPI)? MOSI: 13;
        _ss = (_spi_num == VSPI)? SS: 15;

```
The default is 12,13,14,15 for the VSPI mode, like mine. I understood. It should be in the call line
MiniWebRadio.ino call to do do without parameters

You
`740 tft.begin (TFT_CS, TFT_DC, SPI_MOSI, SPI_MISO, SPI_SCK, TFT_BL); // Init TFT interface`

I will have `740 tft.begin (); // Init TFT interface`

I'll try to accomplish my goal. Thanks for the help.

closed 2019-10-15 by stale[bot]

AVRsteffen, 2022-02-07:
#26
alansbor wrote:
> The default is 12,13,14,15 for the VSPI mode

`    if (sck == -1 && miso == -1 && mosi == -1 && ss == -1) {         _sck = (_spi_num == VSPI)? SCK: 14;         _miso = (_spi_num == VSPI)? MISO: 12;         _mosi = (_spi_num == VSPI)? MOSI: 13;         _ss = (_spi_num == VSPI)? SS: 15; 
`
I am not firm in 'C' or 'C++' but a bit in Verilog. And in this Verilog language mean this 
`_miso = (_spi_num == VSPI)? MISO: 12; `
expressive
"if _spi_num equal to VSPI than _miso = MISO 
else _miso = 12"

I have massive Problems to use GPIO21 + GPIO22 for TFT_CS and TFT_DC together with SPI_SCK=18, SPI_MISO=19, SPI_MOSI=23!
The TFT clocked with 40MHz don't work.

But if I use - and this is the Only condition for work with 40MHz - TFT_CS=13 and TFT_DC=12

ESP32 is a Mystery!