schreibfaul1/ESP32-MiniWebRadio #28: Immer wieder Fehler nach dem compilieren

hbulf, 2019-08-26:
Hallo,
erst mal vielen dank für das super Projekt, allerdings bekomme ich nach dem kompilieren und reset diesen Fehler.

ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Fatal exception (29): StoreProhibited
epc1=0x4008033c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000008fb, depc=0x00000000
Können sie mir da weiterhelfen?

schreibfaul1, 2019-08-26:
Könnte ein Speicherproblem sein, vielleicht ist die Partitionstabelle nicht richtig eingebunden, das sollte so aussehen
[image]

hbulf, 2019-08-27:
Hallo Schreibfaul,
konnte das Problem lösen hatte einen Fehler in der Boards.txt allerdings komme ich mit den wirings nicht ganz klar ist es vieleicht möglich das nochmal bissl konkreter darstellen was mit welchem Kontakt verbunden wird. Vielen Dank schonmal im Vorraus

schreibfaul1, 2019-08-27:
Hallo hbulf,
eine gute Methode ist es alle Komponenten einzeln zu testem. Funktioniert das VS1053 Modul, das Display, der Touchscreen, die SD Karte...? Für jede Bibliothek gibt es Beispielcodes. Am besten ist der Aufbau auf einem Steckbrett und dann wirklich alle Teile einzeln testen. Dann sieht man am besten wo es hakt. Und erst dann alles zusammenbauen. Im Repository gibt es auch einen Stromlaufplan für die von mir benutzten GPIOs. Gerade bei den Displays gibt es verschiedene Varianten, mal ist SD, TFT und Touch auf gemeinsamen SPI-Pins, mal wieder nicht. Wenn so vorgegangen wird bleibt der Erfolg sicher nicht aus.

hbulf, 2019-08-27:
Vielen Dank für deinen Support werde testen und mich ggf. nochmal melden

schreibfaul1, 2019-08-28:
nein, MISO, MOSI und SCK gibt es nur einmal für alles, nur die CS Leitungen sind einzeln verbunden

hbulf, 2019-08-28:
Hallo nochmal,
habe alles getestet und einzeln funtioniert alles wie es soll, nur zusammen will es nicht funktionieren.

ts Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
setup      : Init SD card
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
tft_info   : TFT_CS:22 TFT_DC:21 TFT_BL:17 TFT_MOSI:23 TFT_MISO:19 TFT_SCK:18
tft_info   : init ILI9341
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][Preferences.cpp:354] getUInt(): nvs_get_u32 fail: brightness NOT_FOUND
setup       : SD card not found
 bin mir fast sicher das ich irgendwo was falsch gesteckt habe.

wäre sehr dankbar wenn sie mir nochmal auf die Sprünge helfen könnten

schreibfaul1, 2019-08-28:
Ich vermute einen Fehler auf dem SPI Bus. Für die Komponente, die gerade senden bzw. empfangen soll wird die entsprechende CS-Leitung auf Low gelegt. Es könnte eine CS-Leitung falsch oder nicht verbunden sein. Die Ausgabe sollte in etwa so aussehen:
[image]

hbulf, 2019-08-28:
vielen Dank für die schnelle Antwort hatte alles was Mosi Miso und SCK ist auf jeweils eine Leitung gelegt könnte da der Fehler leigen?
Werde das mal auftröseln und weiter testen.

Danke nochmal

hbulf, 2019-08-29:
Hallo sorry wenn ich nochmal störe habe jetzt nochmal ales hin und her gestekt leider ohne Erfolg.
Ist es möglich das es vielleicht an der SD Karte selbst liegt? Nutze derzeit einen Adapter weil ich nur eine Micro SD hatte.

schreibfaul1, 2019-08-29:
Der Adapter sollte kein Problem sein.
Bei manchen Displays sind alle SPI-Pins einzeln herausgeführt und müssen erst verbunden werden (MOSI von SD mit MOSI vom TP und mit MOSI vom TFT, dgl.  MISO und SCL)

hbulf, 2019-08-31:
Hallo und sorry das ich nochmal nerve bin mittlerweile bischen weiter gekommen.
Brauche aber wohl nochmal deine Hilfe.
WiFI_info  : Connecting WiFi...
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 2 - AUTH_EXPIRE
WiFI_info  : WiFi connected
WiFI_info  : IP address 192.168.179.20
WiFI_info  : connected to FRITZ!Box Gastzugang
rtime_info : Initializing SNTP
rtime_info : Waiting for system time to be set... (1/10)
rtime_info : Waiting for system time to be set... (2/10)
rtime_info : Waiting for system time to be set... (3/10)
rtime_info : Waiting for system time to be set... (4/10)
rtime_info : Waiting for system time to be set... (5/10)
rtime_info : Waiting for system time to be set... (6/10)
rtime_info : Waiting for system time to be set... (7/10)
rtime_info : Waiting for system time to be set... (8/10)
rtime_info : Waiting for system time to be set... (9/10)
rtime_info : Thu Jan  1 01:00:25 1970
vs1053_info: Song stopped incorrectly!
vs1053_info: REG         Contents   bin   hex
vs1053_info: ----------- ---------------- ---
vs1053_info: MODE        1111111111111111 FFFF
vs1053_info: STATUS      1111111111111111 FFFF
vs1053_info: BASS        1111111111111111 FFFF
vs1053_info: CLOCKF      1111111111111111 FFFF
vs1053_info: DECODE_TIME 1111111111111111 FFFF
vs1053_info: AUDATA      1111111111111111 FFFF
vs1053_info: WRAM        1111111111111111 FFFF
vs1053_info: WRAMADDR    1111111111111111 FFFF
vs1053_info: HDAT0       1111111111111111 FFFF
vs1053_info: HDAT1       1111111111111111 FFFF
vs1053_info: AIADDR      1111111111111111 FFFF
vs1053_info: VOL         1111111111111111 FFFF
vs1053_info: AICTRL0     1111111111111111 FFFF
vs1053_info: AICTRL1     1111111111111111 FFFF
vs1053_info: AICTRL2     1111111111111111 FFFF
vs1053_info: AICTRL3     1111111111111111 FFFF
vs1053_info: Connect to new host: 0nline-disco.de:8000
vs1053_info: Connect to 0nline-disco.de on port 8000, extension /
[E][WiFiClient.cpp:213] connect(): lwip_connect_r: 104
vs1053_info: Request 0nline-disco.de:8000 failed!
tft_info   : Total render time was: 29ms
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: info_001 NOT_FOUND
danach geht nichts weiter und auf dem Display erscheint auch nichts

schreibfaul1, 2019-08-31:
Das sieht schon besser aus. Der Fritzbox Gastzugang wird den UDP Port123 zum Zeitserver sperren.

hbulf, 2019-09-01:
Hallo und als erstes größte Hochachtung vor ihrer Geduld mit mir Radio funktioniert jetzt dank der Hilfe.
Allerdings zeigt das Display nichts an Backlight ist an aber sonst nichts wo könnte da das Problem liegen?

hbulf, 2019-09-01:
OK habs selbst gefunden hatte trotz ständiger Kontrolle was falsch gesteckt.
Funktioniert jetzt perfekt nochmals vielen Dank für die Hilfe und Geduld

schreibfaul1, 2019-09-01:
sehr gut!
Über die Weboberfläche kann nun eine eigene Senderliste erstellt werden. Falls ein Logo fehlt kann über die Google-Bildersuche ein passenden Bild ausgewählt werden. Dann das Bild im neuen Tab öffnen, die URL herauskopieren und als LogoUrl einfügen. Mit einem Mausclick wird das Bild übernommen und auf die passende Größe skaliert.
vG sf1

closed 2019-09-01 by schreibfaul1