schreibfaul1/ESP32-MiniWebRadio #31: SD Karte arbeitet nicht

frbr13, 2020-05-08:
Hallo Wolle,

ich habe ein Problem mit der SC Karte da kommt immer ein Fehler beim Start und da bleibt es dann hängen.
Ich habe ein etwas anderes ESP32 als in deinen Bildern.
Espressif ESP32 WLAN Dev Kit Board Development Bluetooth Wifi v1 WROOM32 NodeMCU
ich habe hier die pinkonfiguration gewählt und meine verdrahtung 3 mal geprüft (und auch das Waveshare 2.8 Tft Shield auf einen Arduino uno getestet)

Hast du noch eine Idee ?
VG Frank

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
setup      : Init SD card
tft_info   : TFT_CS:22 TFT_DC:21 TFT_BL:33 TFT_MOSI:23 TFT_MISO:19 TFT_SCK:18
tft_info   : init HX8347D
setup       : SD card not found

schreibfaul1, 2020-05-10:
Hallo Frank,
Display, Tochpad, VS1053 und SD Karte liegen auf dem gleichen Bus. CS->low bestimmt, welche Komponente aktiv ist. Ich gehe von einem Verdrahtungs/Kontaktproblem aus. Jede Komponente kann einzeln auf Funktion getestet werden, dazu gibt es in den Libraries Beispiele zum testen. Wenn einzeln alles funktioniert klappt auch das Zusammenspiel. Ich wünsch dazu viel Erfolg...
P.S. beim Waveshare-Display gibt es einen gelben Klotz, davor können 3 Brücken eingelegt werden damit die Leitungen zur Stiftleiste durchverbunden sind https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9341-HX8347D/blob/master/additional%20info/Waveshare%202.8%20Display.jpg

frbr13, 2020-05-10:
Hallo Wolle,

 

super Danke für deine Hilfe, der Tip mit den 3 Lötbrücken hat geholfen 😊 . Es kam sofort ein Bild auf dem TFT und ich konnte den Sound testen mit den Test Files auf der SD Karte . Jetzt habe ich nur noch das Problem einen Sender zu finden der auch Streamt, ich hatte 5 deiner angelegten  getestet und vor ein paar Tagen Radio Wuppertal (mein Lokaler Sender 😊) hinzugefügt , aber hier kommt immer die Meldung das der Sender die Sender nicht streamen .

 

Danke noch mal für deine schnelle Hilfe 

Schönes Rest Wochenende noch und bleib Gesund

Viele Grüße

Frank

schreibfaul1, 2020-05-10:
das hört sich gut an, die Verkabelung sollte nun in Ordnung sein. Wird die Zeit auf dem Display korrekt angezeigt? Vielleicht besteht keine Verbibdung ins Internet (nur lokal, nach außen wird von Router gesperrt)? Oder mit dem Browser auf die Radio-IP zugreifen und die Stream-URL manuell eingeben.
das Log sollte ähnlich sein zu:
```
WiFI_info  : WiFi connected
WiFI_info  : IP address 192.168.178.9
WiFI_info  : connected to Wolles-FRITZBOX
rtime_info : Initializing SNTP
rtime_info : Waiting for system time to be set... (1/10)
rtime_info : Sun May 10 20:36:57 2020
vs1053_info: Song stopped correctly after 0 msec
vs1053_info: Connect to new host: 0n-80s.radionetz.de:8000/0n-70s.mp3
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
vs1053_info: Connect to 0n-80s.radionetz.de on port 8000, extension /0n-70s.mp3
vs1053_info: Connected to server
vs1053_info: Server: Icecast acc-0.9.10.25 2.4.3-backportCVE-2018-18820
vs1053_info: audio/mpeg seen.
vs1053_info: Cache-Control: no-cache
vs1053_info: Expires: Mon, 26 Jul 1997 05:00:00 GMT
vs1053_info: Pragma: no-cache
vs1053_info: Access-Control-Allow-Origin: *
vs1053_info: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
vs1053_info: icy-description:Von Disco ber Reggae und Schlager bis hin zum Rock. Das sind die Hits der 70er.
vs1053_info: icy-genre:70s, 70er, Oldies, Pop
vs1053_info: icy-name=- 0 N - 70s on Radio
vs1053_info: icy-private:0
vs1053_info: icy-pub:1
Homepage   : http://www.0nradio.com
vs1053_info: Switch to DATA, bitrate is 128, metaint is 16000
lastURL    : 0n-80s.radionetz.de:8000/0n-70s.mp3
vs1053_info: Metadata block 64 bytes
vs1053_info: StreamTitle=Wings - Listen To What The Man Said 
````

frbr13, 2020-05-11:
Heute klappt alles es lag am Internett (wir hatten hier Gestern diverse Probleme)

Danke Gruß

Frank

schreibfaul1, 2020-05-12:
sehr gut!

closed 2020-05-12 by schreibfaul1