schreibfaul1/ESP32-MiniWebRadio #33: TFT causes compile error and Sloeber instruction not up to date

donnersm, 2020-12-06:
Hi,
In the Sloeber instruction you ask to change the TFT.h, removing the line '#include "fonts/Times_New_Roman.h"'
However when compiling, this will give error because in the main sketch, you are refering to it  on line 405 until 410

Compiling also gives an error on the TFT:
..\MiniWebRadio.ino: In function 'void setup()':
..\MiniWebRadio.ino:668:66: error: no matching function for call to 'TFT::begin(int, int, int, int, int, int)'
     tft.begin(TFT_CS, TFT_DC, SPI_MOSI, SPI_MISO, SPI_SCK, TFT_BL);    // Init TFT interface

And

D:\sloeber\arduinoPlugin\libraries\TFT\src/tft.h:91:19: note:   candidate expects 5 arguments, 6 provided
In file included from ..\sloeber.ino.cpp:87:0:

I checked the code on that line but it has only 5 arguments not 6...so not sure what's going on here?

donnersm, 2020-12-07:
Turns out the TFT issues was caused by the wrong library bound..... Following your instructions, at some point I downloaded the wrong library. Not sure it its correct in the instruction or not..

schreibfaul1, 2020-12-07:
The correct library is: https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9341-HX8347D

closed 2020-12-07 by donnersm