schreibfaul1/ESP32-MiniWebRadio #42: compile problem

Beppi4U, 2021-06-16:
Try to compile but ..
html.h: No such file or directory!
...but the directory and scr file exits under MiniWebRadio folder.
Does it has to go anywhere else?
Greetings

PS.What is the reason for the some of  Cyrillic comments? Is this creek language?

schreibfaul1, 2021-06-17:
You can put the html.h and html.cpp in the same folder as the .ino, or try
#include "html/src/html.h"

The fonts should cover all characters commonly used in Europe, including Greek, Russian, Spanish ...

Beppi4U, 2021-07-25:
Hallo Wolle!
Hatte mal wieder Zeit etwas aus zu probieren! 
Irgendwie komme ich mit VS_Code nicht klar! Möchte ich später gerne noch mal eine Chance geben. Bei der Arduino IDE spuckt der Compiler Fehler aus die ich nicht nachvollziehen kann! 
Kannst Du helfen?
Vielen Dank schon mal und Gruß 
Andreas 

[image]

schreibfaul1, 2021-07-25:
Hi,
die Warnung "deprecated sntp.h" kann ignoriert werden, das funktioniert trotzdem. Werde ich noch anpassen.
Es sieht aus, als wenn die TFT Lib fehlt oder eine falsche installiert wurde.  Bitte https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9341-HX8347D als zip runterladen und in die Arduino IDE installieren über Sketch/IncludeLibrary/AddZipLibrary.
Der Rest sieht gut aus.

In VisualStudioCode geht es einfacher nachdem PlatformIO als Erweiterung installiert wurde. Da reicht "Repository klonen" und dort die URL https://github.com/schreibfaul1/ESP32-MiniWebRadio einzutragen. PlatformIO erkennt über die platform.ini den Microcontroller, lädt die richtige ESP32 Arduino Version und alle nötigen Bibliotheken. Und dann compile und upload wie dei der Arduino IDE, fertig.
vG Wolle

SycoJ, 2021-07-26:
Für Arduino IDE reicht es auch aus, wenn du dir die beiden Dateien und den Ordner Fonts aus: ESP32-TFT-Library-ILI9341-HX8347D/src in den Ordner kopierst, wo deine Miniwebradio.ino liegt. 

Da brauchst du die ZIP nicht einbinden. Wolle hat ja wunderbar über include die Dateien eingebunden.

Hat bei mir wunderbar geklappt.

Beppi4U, 2021-07-26:
Hi!
Jetzt wird zwar kompiliert(Arduino) aber es scheint als kenne die IDE das Partitionsschema nicht! 
In den Boards Definitionen (.txt) habe ich das Schema eingetragen kann es aber nicht auswählen! Auch No Ota Huge APP/NoOTA 1mb SPiffs geht nicht!
Ich habe einen bösen Verdacht......liegts an dem Board selber?? 
....hab ein ESP32 Wroom-32U zum Aufbau genommen.
Gibt es hier jemanden der mal in einer Skype runde etwas helfen könnte??
.......ich weiß das das eher ungewöhnlich ist aber Fragen kann ich ja mal!
Danke und Gruß
Andreas

[image]

Beppi4U, 2021-07-26:
[image]

schreibfaul1, 2021-07-26:
Soweit ganz gut!. Bei Visual Studio Code geht vieles automatisch. In der Arduino IDE muss halt einiges von Hand gemacht werden. Je nach Betriebssystem gibt es irgendwo den Ordner .arduino15/packages/esp32/hardware/esp32/1.0.6/tools/partitions
Und in den "partitions" Ordner muss die miniwebradio.csv kopiert werden. Das ist notwendig, weil bei den angebotenen Partitionstabellen keine mit ausreichend nvs dabei ist.
vG Wolle

Beppi4U, 2021-07-27:
Hi! Also unter esp32/1.04/tools/partitions liegt das .csv (siehe Bild).
lässt sich dennoch nicht kompilieren! Ist die ESP32 Version zu klein? Du hattest immer V1.06. 
vG Andi

[image]

schreibfaul1, 2021-07-27:
Die csv ist an der richtigen Stelle. Ich benutze V1.0.6 aber V1.0.4 dürfte auch funktionieren. 
[image]
Und ich kann auch das Partitionsschema auswählen. Vielleicht hast du die drei zusätzlichen Zeilen in boards.txt an der falschen Stelle?. Das muss zu dem ausgewählten Board passen. In der Arduino IDE ist eben viel Handarbeit nötig. Der Sketch Ordner sieht so aus:
[image]
und den Inhalt der main.cpp in die MiniWebRadio.ino kopieren oder die main.cpp umbenennen. Hoffe das hilft :-)

Beppi4U, 2021-07-31:
Also ich hab jetzt mehrere Versuche mit ArduIDE und mit PlatformIO hinter mir und keine hat zum Erfolg geführt! Platform scheint an einem Punkt nicht mit dem Speicherkontroller klar zu kommen! Auch hier gehe ich stark davon aus das das Problem der ESP ist und ich einfach ein falsches Board genommen habe! Ich werde ein neues Bestellen und dann sehen wir weiter.....ich will das hinbekommen!! (bin aber erstmal ne Woche in Urlaub....kann also noch etwas dauern!)
Oder jemand hat hier die erlösende Lösung parat!
An die Mit/Leser: Welche Board's haben denn schon bei Euch funktioniert??

Mein Board ist ein ESP32 Wroom - 32U mit ausgeführtem WIFI Antennenstecker.
vG Andi

[image]

[image]

[image]

SycoJ, 2021-08-04:
Hallo Andi,

Ich hab mit so einem Board einwandfrei in ArduinoIDE kompiliert:

ESP32 DevKit V4 mit Espressif ESP32 WROOM 32u

[image]

Ich hab unten links GND, wo du cmd auf deinem Board hast. 

Erstmal schönen Urlaub Dir :)

utw3v0, 2021-08-04:
Ändere mal das board in der PlatformIO.ini von "board = esp32dev" zu board = "esp32doit-devkit-v1" oder zu "nodemcu-32s", hab das aus irgendwelchen Gründen auch geändert, weiß nicht mehr was es war aber das passt jedenfalls besser zu deiner MCU, viel Glück. 

Gruß 3v0

Beppi4U, 2021-09-10:
Hallo! 
Ich hatte mal wieder etwas Zeit an dem WebRadio zu testen! 
Ich habe es mittlerweile geschafft richtig unter Arduino IDE zu Kompilieren.
Ich habe nach dem ersten Start gelesen das er die SD Karte wohl nicht gefunden hatte . War ein Mini SD Karte und ich habe die Befürchtung das einige Kartenleser damit nicht zurechtkommen. Also Andere SD Karte genommen Inhalt draufkopiert. SD wurde dann erkannt.
Allerdings bleibt das Display dunkel und via COM Port wird folgende Fehlermeldung ausgegeben!
Hat jemand einen Tipp was ich falsch gemacht habe und was die Meldungen bedeuten?
Gruß
Andreas

[image]

SycoJ, 2022-01-13:
Hallo Andreas, 

Station sollte eine CSV Datei sein und keine TXT-Datei. Lade dir mal bitte Wolles aktuellste Dateien herunter. Dein Fehler sagt ja, dass er die "Stations.txt" nicht gefunden hat. Falls es vor der CSV eine TXT gewesen ist, muss die im selben Ordner liegen. Nämlich unter "Content on SD Card". Oder hat sich dein Problem mittlerweile gelöst?

closed 2022-04-28 by stale[bot]