schreibfaul1/ESP32-MiniWebRadio #43: WaveShare 2.8 und CATALEX MicroSD Card Adapter

SycoJ, 2021-07-03:
Hallo Wolle,

vielen Dank vorab für dein super Projekt!

Ich habe mir das 2,8 Zoll Touch Display von WaveShare ohne SD-Slot bestellt. (Ich wusste nicht, dass es das auch ohne gibt) 
Kann ich über den MicroSD Card Adpter die Bilder etc. auf dem Display herstellen oder benötige zwingend das Display mit SD-Slot auf der Rückseite?

Wenn ja, kannst Du mir sagen, an welche GPIOS ich ran muss oder verhält sich das aquivalent zur MP3 Variante mit deinen Zeichnungen und Verbindungsübersichten?

Und ist es richtig, dass beim Display die 5V nicht angeschlossen werden ? (Siehe deiner Schematic)

Vielen Dank im Voraus,

Greg

schreibfaul1, 2021-07-05:
Hallo Greg,
du hast einen Fehler im Schaltplan gefunden. Das Display braucht natürlich die +5V, das werde ich demnächst korrigieren. Wenn das Display keinen Kartenslot hat kannst du natürlich einen externen SD Card Adapter verwenden. Nur der Catalex ist nicht empfehlenswert weil er kein weiteres Gerät am SPI Bus duldet. In Additional_Info Ordner gibt es einen Umbauvorschlag für den Catalex.
Mein Vorschlag wäre dieser Adapter.
[image]
oder ein Selbstbau wie hier: https://alexlubbock.com/micro-sd-adapter-esp8266-esp32

vG
Wolle

SycoJ, 2021-07-06:
Vielen Dank für deine schnelle Rückmeldung:)

Ja beim Umbau des Catalex hattest du den R1 mit Pin 13 verbunden soweit ich mich erinnere. Schau dann nochmal nach. 

Die 5v sind ja logisch aber der Vollständigkeit halber ^^ 
Hattest du ursprünglich auch das Waveshare ohne SD im Einsatz? Ich meine das im Repositories gesehen zu haben.

Danke für die Alternative bzw den DIY Bauvorschlag. Letzteres ist eigentlich so simpel, dass man nicht drauf kommt^^

Hab vielen Dank.

LG Greg

closed 2021-07-08 by SycoJ