schreibfaul1/ESP32-MiniWebRadio #44: Aktuelle Software fehlerfrei?

Accuvue, 2021-07-06:
Hallo Wolle,
Ich verfolge Dein Projekt seit Längerem. Ich habe es zunächst wie beschrieben aufgebaut, habe mir aber eine modifizierte Variante draus gebastelt. Mit kleineren 1,77" Display, Kurzhubtasten und so klein, dass es inklusive Netzteil in eine UP-Schalterdose passt. Dein Programm habe ich entsprechend angepasst. Vieles übernommen. Manches habe ich, finde ich, besser gemacht. Beispielsweise habe ich einen "Fader" reingebaut, damit die Umschaltung der Sender etwas gefälligst klingt. Eine OTA-Downloadmöglichkeit auch. Aber das sind nur Kleinigkeiten. 

Ich bin immer gespannt auf Deine Updates. Besonders das Vorletzte brachte bezüglich der HTML-Stabilität echt viel. Das ist immer etwas aufwändig, um alle Deine Änderungen nachzuvollziehen, aber bisher war es immer machbar. 
Nicht ganz zufrieden bin ich mit dem aktuellen Versionsstand. Vorausgesetzt ich habe es richtig in meine Software eingepflegt, läuft die Webseite relativ instabil. Bis hin zum Absturz der Webseite. Ist das bei Deiner Hardware auch so? Arbeitest Du da an einem Update?

Hendrik

schreibfaul1, 2021-07-06:
Hallo Hendrik,
danke für den Kommentar. Ich habe in der letzten Zeit viel geändert, manches war nicht mehr zeitgemäß. Der verwendete CORS Proxy funktionerte nicht mehr, wie auch der Timer-Interrupt in V1.0.6, das macht jetzt der Ticker. In Sachen HTML habe ich ein Websocket integriert, damit vom ESP zum Browser Daten gesendet werden können ohne die sekündlichen XHR Anfragen zu nutzen. Eigentlich wäre der ESP32 AsyncWebServer dafür die erste Wahl, aber leider gibt es Probleme beim Laden meherer Grafikdateien das schafft die Lib nicht wirklich gut. Daher meine Eigenentwicklung. Die "html"-Dateien habe ich überarbeitet und in "websrv" umbenannt. Das ist seit Websockets auch nicht mehr reines HTML. Und es funktioniert bei mir stabil, ich hoffe bei dir dann auch :-))
Die SW ist nur eine Vorlage für eigene Entwicklungen. Daher freut es mich wenn eigene Erweiterungen und Änderungen programmiert werden. Ich habe einen Blog gefunden, wo das Radio in eine Wand eingebaut wurde, das könnte Dein Projekt sein? das finde ich prima! Der "Fader" ist auch eine gute Idee, kommt vielleicht demnächst bei mir auch. 

vG Wolle

Accuvue, 2021-07-06:
Hallo Wolle,
ja, ich hatte mal ein paar Bilder von meinem Radio eingestellt. Die wirst Du sicherlich gesehen haben.

Deine aktuellen Änderungen werde ich mir mal ansehen. Mal schauen, ob ich sie in mein Projekt integrieren kann. Ansonsten lasse ich den vorherigen Stand (die mit den sekündlichen XHR Anfragen). Die läuft wirklich ziemlich stabil.

Ich werde Dein Webradio weiter verfolgen. Hast Du noch Pläne für grundlegende Erweiterungen?

Hendrik

closed 2021-09-19 by stale[bot]