schreibfaul1/ESP32-MiniWebRadio #45: Laggy

SycoJ, 2021-07-08:
Hi Wolle, 

hoffe du kannst noch mal helfen. Der Ton wird unterbrochen sobald ich eine Taste auf dem Touch drücke.

Das NT, was ich verwende gibt 1,7 A bei 5V aus. Das sollte passen. Der PSRam ist ja Standardmäßig aktiviert. 
Kann es an der Kabellänge liegen? Die Kabel sind unter 10cm lang.

Hast du eine Idee woran es liegen könnte?

Vielen Dank im Voraus.

schreibfaul1, 2021-07-10:
Tonaussetzer lassen sich nicht immer ganz vermeiden. Der VS1053 kann nicht viel zwischenspeichern und muss ständig mit Audiodaten beliefert werden. Das auch das Display auf dem SPI liegt wird das nicht immer sichergestellt, zumal der Grafikaufbau Rechenzeit benötigt. Du kannst den Effekt minimieren, indem die  SPI Taktfrequenz für die Geräte so hoch wie möglich gesetzt wird.

SycoJ, 2021-07-10:
Vielen Dank.

Finde ich die in der TFT.cpp und ist Standard auf 40mhz gesetzt?

[image]

Vielen Dank im Voraus,

Greg

schreibfaul1, 2021-07-10:
40MHz für TFT ist voreingestellt.
Die neueren SD Karten schaffen 80MHz.
Ich glaube eine gute Lösung wäre es einen eigenen Task für vs1053.loop() mit hoher Priorität zu erzeugen. Dieser könnte über eine Queue gesteuert werden. Das werde ich in den nächsten Tagen testen. Wichtig ist nur, das der VS1053 regelmäßig seine 32 Byte Audiodaten erhält.

SycoJ, 2021-07-11:
Hi Wolle, das wäre klasse, wenn du es testen könntest. Ich muss meine microsd Karte mal austauschen, die list wohl nur mit 2 mb/s. Sollte dennoch ausreichen aber ich teste es die kommende Woche, wenn sie da ist. 

Bei mir äußert sich das ruckeln so:

Uploading 20210711_200349_1.mp4…

Bei dir ist das im Video jedoch nicht der Fall.

SycoJ, 2021-07-19:
Hi Wolle, 

hattest du ggf. die Möglichkeit zum Testen? 

Eine neue MicroSD hat, wie vorher editiert, leider keinen Erfolg gebracht.

Vielen Dank im Voraus

schreibfaul1, 2021-07-19:
Hallo,
ganz so viele Tonaussetzer habe ich nicht. Während ein Icon auf den Display gezeichnet wird muss der VS1053 auf seinen internen Speicher zurückgreifen. Ich habe in der VS1053_ext lib etwas verändert, um den Bitstreambuffer der VS1053 besser zu füllen. Vielleicht hilft das schon etwas.

https://user-images.githubusercontent.com/26044260/126216489-2d2075fb-70a1-4e16-bd94-5d1403143815.mp4

SycoJ, 2021-07-19:
Es hilft wirklich etwas. Vielen Dank.

Da es jedoch nicht mal annähernd deiner Performance entspricht, muss leider einen Hardware defekt vermuten und werde deshalb ein neues VS1053 Modul bestellen.

Ich gebe Bescheid sobald es da ist.

Vielen lieben Dank erstmal für deinen Support.

LG Greg

SycoJ, 2021-07-26:
Habe jetzt das neue VS1053 von nem anderen Händler testen können. Der Effekt bleibt bestehen.

Den Esp32 habe ich auch schon getauscht.

Daher bestelle ich jetzt noch ein anderes Display, um die Hardware komplett ausschließen zu können.

Wenn es wirklich am Display liegt, schreibe ich diesen Fehler auch auf Englisch damit es jeder nachvollziehen kann und keinen weiteren Fall öffnet ;)

Bis dahin erstmal allen einen guten Start in die Woche

MovingShadow101, 2021-07-26:
Hi Wolle,
erstmal Danke für dieses klasse Projekt.
Habe meins jetzt gerade fertiggestellt und auch mit Tonaussetzern und einem leisen hochfrequenten Piepen aus den Lautsprechern zu kämpfen gehabt.
Mit Kopfhörern direkt am VS1053 waren bei mir alle Probleme weg.
Hab jetzt EI14/LT700 Audio Transformatoren zwischen dem VS Modul und meinem PAM Verstärkermodul gesetzt
und so eine galvanische Trennung des Audiosignales erreicht.
Ergebnis: keine Tonaussetzer und ein guter Klang.
Vielleicht kann es hier ja auch jemanden weiterhelfen.

greetings
[image]

schreibfaul1, 2021-07-26:
Das ist Klasse! Tolles Gerät und prima Aufbau!
Ich habe heute die vs1053_ext Lib aktualisiert. Es gibt ein neues AudioBuffer Konzept. Die Chunk- und Metadaten werden jetzt vorher abgefangen und verarbeitet. Das bring noch etwas an Performance für den VS1053. Der BitStreamBuffer wird besser gefüllt. Zusätzlich wird nun automatisch das FLAC Plugin geladen. Es werden Webfiles zuverlässig erkannt. WebStreams laufen 24/7, für Audidateien aus dem Netz gibt es dagegen eine zeitliche Begrenzung. Webfiles sind z.B. Podcasts, das können aber auch Dateien sein, die auf einen eigenen Server oder der Fritzbox liegen (m4a, mp3, wav oder flac).
vG Wolle

closed 2021-07-26 by schreibfaul1

SycoJ, 2021-07-27:
@MovingShadow101 

Du hattest dann eine Rückkopplung. Ich hatte auch einen externen Verstärker mit eigener Stromversorgung. Das hat jedoch die Aussetzer nicht behoben.

Und schönes Design. Schaut sehr schön aus :)Bin auch noch am Designen und hoffe es bald auf den 3D Drucker bringen zu können:)

@Wolle: 

Ich hatte vorher noch deine Audio.h eingebunden und nun die neue vs1053_ext.h verwendet. Das funktioniert nicht, da "AudioBuffer" zweimal definiert ist und es kommt zum Kompilierungsfehler.

Einfach die Audio.h  und Audio.cpp und in der Miniwebradio.ino include Audio.h entfernen. Dann funktioniert alles. Dies nur falls ein User mal den Fehler haben sollte. :)

Und vielen Dank für deine Optimierung :)

schreibfaul1, 2021-07-28:
Hi SycoJ,
danke für deinen Kommentar. Ich habe tatsächlich viele Routinen meiner audioI2S Lib hierher kopiert, damit beide Bibliotheken etwa den gleichen Stand haben. Und den alten Ringbuffer habe ich durch den AudioBuffer ersetzt. Das bringt noch etwas an Zuverlässigkeit. Zuden kann der AudioBuffer mit PSRAM arbeiten. Der Nachteil ist allerdings, dass das PSRAM gegenüber dem SRAM bei größeren Datentransfer (intern ist nach 12K? die Queue voll) die Geschwindigkeit absinkt. Meistens ist das nicht zu bemerken jedoch habe ich einen Stream gefunden der dann Tonaussetzer produziert. Muss man halt abwägen, ob man PSRAM ativieren will.

vG Wolle

AVRsteffen, 2022-01-26:
Super tolles Projekt Wolfgang! Danke Dafür! Eines der Besten, die ich bis jetzt gesehen habe. 

Die Selben Aussetzer wie SycoJ hatte ich auch erst. Ich hatte dann versucht die SPI Frequenz für die SD.begin(...) in der .ino hoch zu setzen. Diese ist dort auf 8.000.000 Mhz gesetzt. Dies führte zu keinem Erfolg. Ich machte dazu Tests mit einem Logic Analizer. Dabei fand ich heraus:
SPI CLK SD = 4.000.000 Mhz
SPI CLK TFT = 40.000.000 Mhz
SPI CLK VS1053 = 6.667.000 Mhz
Also bin ich auf der Suche nach der niedrigen SD CLK auf die SD.h gestoßen. Da ist die Frequenz fest auf 4.000.000 Mhz für die SD card eingetragen. Leider konnte ich jetzt meine SD-Card mit max. 16.000.000 Mhz ansteuern bevor sie nicht mehr zuverlässig funktioniert und irgendwann mal aussteigt und sich dann selber blockiert. Und dabei hab ich schon eine super schnelle Karte von SanDisc. Eine SanDisc Edge 4GB SDHC. Die sollte 80MB/s schaffen und explizit für hohe Datentransfairaufkommen sein.

Die Aussetzer beim Laden der Buttons ist zwar deutlich weniger geworden. Dennoch immer noch vorhanden.

Vielleicht könnte man die SPI Kommunikation auf 2 SPI Busse aufteilen. Vielleicht sogar mit DMA zum VS1053 schaufeln. So sollten keine Aussetzer mehr entstehen. Und den zweiten SPI zur Kommunikation zwischen
TFT <-HSPI->ESP32<-HSPI->SDcard
ESP32 <-VSPI-> VS1053

Besteht diese Möglichkeit?