schreibfaul1/ESP32-MiniWebRadio #46: Ili9341 Display integrierter SD slot + externe SD

SycoJ, 2021-07-19:
Hi Wolle,

Ich brauch leider nochmal deine Hilfe.

Ich habe das Ili9341 mit integriertem SD Slot und habe extern einen SD Kartenadapter angelötet. (Vielen Dank für deinen Tipp: [SD Card Adapter ](https://alexlubbock.com/micro-sd-adapter-esp8266-esp32) )

Ich habe nun die ganzen Dateien aus ESP32-audioI2S eingebunden und in der Webradio.ino #include Audio.h einwandfrei in der Arduino IDE kompiliert.

Er liest jedoch die externe SD nicht. Hab auch schon den Ordner "mp3files" in Content for SD Card gelöscht und auf der externen SD erstellt. Dann finde ich gar keine MP3 files.

In der .ino sucht er ja nur nach dem Ordner "/mp3files"

String listmp3file(const char * dirname="/mp3files", uint8_t levels=2, fs::FS &fs=SD){ static String SD_outbuf=""; 

Hast du eine Idee, was ich übersehen oder vergessen habe könnte?

Vielen Dank im Voraus Greg

schreibfaul1, 2021-07-19:
Hallo Greg,
wenn das Display schon einen Kartenslot hat dann wäre das doch die erste Wahl. Einfach MISO/MOSI/SCK von Kartenleser mit dem SPI des Displays verbinden und SD_CS mit dem Pin des ESP.
Nach deiner Beschreibung bin ich mir nicht ganz sicher wie die aktuelle Konfiguration ist. Gibt es eine eine externe und interne Karte?
Oder geht nur das Lesen des mp3 Ordners nicht? Dann vermute ich du hast bereits die Version V2.0.0-alpha1. In der neuesten Version wird zwischen file.name() und file.path() unterschieden. Falls das der Fall ist kannst du in listmp3file(...) die Zeile mit dem path aktivieren und die Zeile mit name() auskommentieren.

vG Wolle

SycoJ, 2021-07-19:
Hallo Wolle, 

Ja es gibt eine im Display befindliche Karte und eine externe. Die im Display befindliche Karte hat deine Ordner mit den Icons, Buttons etc. und den mp3 files Ordner. 

Die zweite Karte ist ein microsd Adapter. Also eine SD Karte, wo man eine microsd reinstecken kann. Den Adapter hab ich an den Esp32 gelötet: CS, Mosi, GND, Vcc(3,3V), SCLK und Miso.

Setze ich die SD Karte mit den Button Bildern etc und den mp3files Ordner in das Display ein, werden die Mp3s wunderbar abgespielt. Ich möchte jedoch einen zweiten Slot, wo ich nur die Karte einsetze, um Mp3 Dateien abspielen zu können. 

Das ganze sieht aktuell so aus:

[image]

Die Karte direkt im Display soll alles tun außer MP3 beherbergen. Die Mp3s sollen von der zweiten, externen Karte abgespielt werden.

Vielen Dank im Voraus,

Greg

schreibfaul1, 2021-07-19:
Eine zweite SD Instanz zu eröffnen könnte schwierig sein. Besser ist es die zweite SD Karte als 1wire SD_MMC einzubinden. Und dann für die mp3files "SD" in "SD_MMS" ändern.
String listmp3file(const char * dirname="/mp3files", uint8_t levels=2, fs::FS &fs=_**SD_MMC**_)
Die Pins der SD_MMC lassen sich nicht frei wählen sondern es werden die vordefinierten Pins benutzt. Die SD_MMC ist etwas kompliziert beim Bootvorgang, hier gibt es einige Tipps in der readme: https://github.com/biologist79/ESPuino

SycoJ, 2021-07-20:
Es ist wirklich schwierig ^^

Ich hab den GPIO 2 (VS1053_CS) auf GPIO 13 gelötet und definiert. 

Damit kann ich die fest definierten GPIO (14, 15 und 2) für den SD Karten Adapter verwenden. 

Die Änderung bringt leider keinen Erfolg. 

Hast du ggf. noch eine Idee? Wenn nicht, dann lass ich das Projekt

utw3v0, 2021-07-20:
Du benötigst nur einen einzigen pin für cs alle anderen hast du alle schon mosi miso und clk ... worauf du allerdings achten solltest ist das es pins gibt die man besser nicht definiert weil sie beim booten ne andere funktion haben bspw. Pwm etc ... auch die pins die am internen flash hängen sind nicht geeignet ... steht aber in der doku zum esp ... musste mal googln ....

⁣TypeApp for Android herunterladen ​

SycoJ, 2021-07-21:
Die Verbindung hatte ich schon aufgebaut. 

Ich hab gem. Link: https://github.com/biologist79/ESPuino 
versucht über SD_MMC anzubinden. 

CS pin wird gar nicht im Modus benötigt. 

Oder meinst du einen separaten, zweiten CS nur für die externe SD Karte zu definieren ?

Danke im Voraus,

Greg

schreibfaul1, 2021-07-21:
Hi Greg,
das mit dem CS Pin meinte ich im Zusammenhang mit einer zweiten SD Instanz, also "SD" und "SD1"
SD_MMC funktioniert ohne SPI, ist schneller, hat aber fest zugewiesene Pins.

File mp3file;
File mp3file1;

in setup()
   SD.begin(SD_CS);
   SD_MMC.begin();

und dann
   mp3file = SD.open("Titel1.mp3");
   mp3file1 = SD_MMC.open("Titel2.mp3");

vG Wolle

SycoJ, 2021-07-26:
Hallo, 

Ich bin noch nicht zum Einbinden gekommen. Hab aber gestern mal mit dem ili9486 kompiliert. ^^

Ich gebe Bescheid sobald es Neuigkeiten gibt.

schreibfaul1, 2021-07-28:
Das mit der ILI9486 Lb könnte funktioneren. Dann verschenkt man Platz auf dem Display. Könntest vielleicht die Schrift größer stellen? 
Noch ein Tipp:
[image]
Wenn du das graue Display in der Mitte benutzt (gibts beim großen "A"  zusammen mit einem passenden Gehäuse) hast du echtes SPI und kannst 80MHz einstellen. Die blauen simulieren SPI zu parallel und sind langsamer.