schreibfaul1/ESP32-MiniWebRadio #47: aktueller Code

cadtargo, 2021-08-28:
Hallo Wolle,

der aktuelle Code MiniWebRadio.ino ist scheinbar nicht mehr enthalten. Kannst du das nochmal dazu legen?

Vielen Dank
Gruß Frank

schreibfaul1, 2021-08-28:
Hallo cadtargo,
das Projekt habe ich für die Benutzung mit PlatformIO umgeschrieben. Das ist für die meisten User einfacher zu benutzen. Einfach den Link zum Projekt kopiern, in PlatformIO einfügen, Repository klonen und hochladen reicht. Das Kopieren der miniwebradio.csv und das Editiern der boart.txt entfällt. Und die externen Libraries müssen nicht extra installiert werden.

Falls du weiterhin die Arduino IDE benutzen willst, nenne die main.cpp in MiniWebRadio.ino um.
Es gibt einige Verbesserungen.  Aktualisiere bitte auch die Lib vs1053_ext damit es keine Kompilierungsfehler gibt.

viele Grüße
Wolle

Es gibt einige Verbesserungen

cadtargo, 2021-08-28:
Ok, vielen Dank für die schnelle Antwort. PlatformIO ist mir nicht bekannt, muss ich mich mal mit beschäftigen. Auf die schnelle blicke ich da nicht durch :-) Ich habe es erstmal über die Arduino Ide installiert.
Die Stundenansage muss ich wieder auskommentieren, sonst läufts gut.

Gruß Frank

closed 2022-01-03 by stale[bot]