schreibfaul1/ESP32-MiniWebRadio #48: ESP32-TFT-Library-ILI9486

d3frost, 2021-08-28:
Hallo Wolle,

erstmal VIELEN Dank für dieses geile Radio ! 
Ich habe ein 3,2 Zoll RPI LCD ....... bekomme es aber nicht eingebunden. 
Hast Du einen kleinen Hinweis wie ich das hinbekommen könnte ? 

https://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)

schreibfaul1, 2021-08-28:
Auf der Webseite vom Waveshare steht leider nicht, was für ein Controller verbaut worden ist.
Aber hier http://www.lcdwiki.com/3.2inch_RPi_Display steht ILI9341. Somit wird das sicher mit der ESP32-TFT-Library-ILI9341-HX8347D Lib laufen

d3frost, 2021-08-29:
Hallo, ich habs probiert ... aber leider ohne Erfolg. Meinst Du wenn ich Dir so ein Display mal als Testobjekt schicke ( kannst es auch gerne danach behalten ) bestünde die Chance das Du eine Lösung findest ?

d3frost, 2021-08-29:
Ich habe irgendwie den Verdacht es könne am fehlenden BL des Displays liegen .....

closed 2022-01-03 by stale[bot]