schreibfaul1/ESP32-MiniWebRadio #49: Touch is not working

stsommer, 2021-09-19:


schreibfaul1, 2021-09-19:
Hallo, viele Displays (zB die roten mit ILI9341) haben eigene Pins für MISO/MOSI/SCK. 
Z.B. werden TFT_MOSI(SDI) mit TP_MOSI (T_DIN) und SD_MOSI (SD_MOSI) des Displays miteinander verbunden und dann mit dem definierten Pin Des ESP. Das gleiche mit MISO und SCK.

stsommer, 2021-09-19:
Ja. Genau so ein Display benutze ich. 
Danke für die schnelle Antwort und den Hinweis auf die Anschlüsse. Werde es dann mal probieren und hoffe das es dann klappt. 
Es ist ein wirklich geiles Projekt an dem ich bisher sehr viel lernen konnte. 

Mir ist noch die Idee gekommen das WebRadio mit einem weiteren Wemos D1 Mini Pro auszustatten. Dieser bietet dann einen AP um das ganze etwas portabler zu machen. Denke das der ESP32 mit dem Radio genug zu tun hat. 
Was denkst Du?

stsommer, 2021-09-21:
Hallo Wolle,
habe nun die übrigen Pins am Display auch verdrahtet, und siehe da, es funktioniert. Noch einmal danke für den Tipp.

Nun noch eine andere Frage. Das Display flackert sehr stark bzw werden die Farben teilweise nicht richtig dargestellt. Kann das an dem Aufbau auf dem Breadbord, oder an der vielleicht schlechten Qualität des Breadbord oder der Jumperkabel liegen?

stsommer, 2021-09-28:
Die Probleme sind behoben. Das Flackern lag an der eher schlechten Verkabelung auf dem Breadboard. Mit einer festen Verdrahtung auf einer Platine funktioniert alles einwandfrei.

closed 2021-09-28 by stsommer