schreibfaul1/ESP32-MiniWebRadio #50: Touch ist invertiert

stsommer, 2021-09-28:
Hallo und guten Abend Wolle.

Noch einmal Danke für das tolle Projekt und für Deine Hilfe bis hierher.
Allerdings habe ich noch ein Problem. Der Touch ist invertiert. Heißt links oben ist rechts unten. Habe auch schon in den geschlossenen Issues Dine Tipps gelesen, komme aber irgendwie nicht weiter.
Wie genau muss ich vorgehen um den Touch in die richtige Position zu bekommen??
Danke schon einmal im Voraus.

schreibfaul1, 2021-09-29:
Hallo,
für den Anschluss des TP gibt es scheinbar keine Norm. Jeder Hersteller hat sein eigene Variante.
Im setup() gibt es ` tp.setRotation(3);` probiere mal  `tp.setRotation(1);`

stsommer, 2021-09-29:
Servus Wolle. 
Danke für die schnelle Antwort. Das werde ich Morgen mal ausprobieren und dann berichten ob es geklappt hat. Muss Morgen auch noch die IR Fernbedienung einrichten. Habe wohl andere Codes als die in Deiner Bibliothek. Aber auslesen und umschreiben ist ja schnell erledigt. 

Grüße aus Polen.

stsommer, 2021-10-12:
Hallo Wolle.
Habe heute mal Deinen Tipp ausprobiert und die Variable für tp.setRotation auf 1 gesetzt, und jetzt funktioniert es.
Danke noch einmal.

closed 2021-10-12 by stsommer