schreibfaul1/ESP32-MiniWebRadio #52: Calling function with more parameters

OK1FWG, 2021-11-12:
Hi,
trying to compile latest master code, I've run into this issue:

```
src/main.cpp: In function 'void setup()':
src/main.cpp:657:66: error: no matching function for call to 'TFT::begin(int, int, int, int, int, int)'
     tft.begin(TFT_CS, TFT_DC, SPI_MOSI, SPI_MISO, SPI_SCK, TFT_BL);    // Init TFT interface
                                                                  ^
In file included from src/main.cpp:78:0:
include/tft.h:91:19: note: candidate: void TFT::begin(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)
         void      begin(uint8_t CS=22, uint8_t DC=21, uint8_t MOSI=23, uint8_t MISO=19, uint8_t SCK=18);
                   ^
include/tft.h:91:19: note:   candidate expects 5 arguments, 6 provided
```

Is the latest parameter intended for use to switch the LCD on/off? Where is it handled?

Also,
this part won't compile either:

```
src/main.cpp:168:10: error: no matching function for call to 'TFT::TFT(int)'
 TFT tft(0);                     // parameter:  (0)ILI9341, (1)HX8347D
          ^
In file included from src/main.cpp:78:0:
include/tft.h:88:9: note: candidate: TFT::TFT()
         TFT();
         ^
include/tft.h:88:9: note:   candidate expects 0 arguments, 1 provided
include/tft.h:85:7: note: candidate: constexpr TFT::TFT(const TFT&)
 class TFT : public Print {
       ^
include/tft.h:85:7: note:   no known conversion for argument 1 from 'int' to 'const TFT&'
src/main.cpp: In function 'void setup()':
src/main.cpp:657:66: error: no matching function for call to 'TFT::begin(int, int, int, int, int, int)'
     tft.begin(TFT_CS, TFT_DC, SPI_MOSI, SPI_MISO, SPI_SCK, TFT_BL);    // Init TFT interface
```

I am using PIO.

schreibfaul1, 2021-11-12:
wrong TFT lib?
https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9341-HX8347D

OK1FWG, 2021-11-12:
I don't think so. I am using your library as instructed.
```
|-- <ESP32-IR-Remote-Control> 1.0.10+sha.781d092
|-- <ESP32-TFT-Library-ILI9341-HX8347D> 2.0.0+sha.f95be99
```

```
Compiling .pio\build\esp32dev\src\IR.cpp.o
Compiling .pio\build\esp32dev\src\main.cpp.o
Compiling .pio\build\esp32dev\src\tft.cpp.o
Compiling .pio\build\esp32dev\src\vs1053_ext.cpp.o
```
The header files are in the include directory.

schreibfaul1, 2021-11-12:
I just tried it again, deleted the existing project, copied the Github URL, in PIO clone repository, build, upload, all libraries are reloaded and the compiler doesn't complain

OK1FWG, 2021-11-12:
Did the same and it is working as a charm. Interesting. Thank you. Appreciate your good work and looking forward to assemble the HW soon ;-)

closed 2022-04-17 by stale[bot]