schreibfaul1/ESP32-MiniWebRadio #54: Bluetooth-Streaming

Accuvue, 2021-11-18:
Hallo Wolle,
ist es geplant, in das Projekt noch eine Möglichkeit zum Bluetooth-Streaming zu integrieren? Ist das von den vorhandenen Resourcen überhaupt möglich? 

VG
Hendrik

schreibfaul1, 2021-11-18:
Dazu müsste man das Audiosignal vom VS1053 zurück zum ESP32 führen. Das wäre schon einiges an Bastelarbeit. Zudem benutzt BT und WiFi denselben Frequenzbereich und die gemeinsame Antenne. Das könnte sich gegenseitig beeinflussen.
Im ESPuino Projekt gibt es Überlegungen genau das zu machen, (das Audiosignal im 2x int16_t Format mit 44,1KHz über BT zu streamen) Wir werden sehen, wie erfolgreich das sein wird.
vG Wolle

compactflash, 2023-10-04:
Hi,
Ich wollte das auch schon immer anfragen, nur BLUETOOTH_SINK_MODE wäre für mich ok und müßte doch möglich sein.
VG

schreibfaul1, 2023-10-05:
Hallo @compactflash ,
das wäre sicher eine Sache, die machbar ist.
Weitere Baustellen sind:

- bessere DLNA Integration,  Bedienung über das Display
- Ersetzen des Helix AAC Dekoders durch faad2, damit Parametric Stereo ermöglicht wird
- DMA über SPI,  Grafiken schneller aufs Display schreiben