schreibfaul1/ESP32-MiniWebRadio #55: touchscreen calibration

steevemadison, 2021-11-18:
i'm struggling at my touchscreen , it does work but X is swap , but not Y
my setup is tft_rotation(3) and my tp_rotation(1) , i tried tp_rotation(3) and it's up side down

i load touchpad.ino from the tft driver example the result i get is .
top left x300,y0
top right x0,y0
bottom left x300,y236
bottom right x0,y236

i found at the end of tft.cpp some value , but they don't correspond to my value
any clue on what to edit ???

thank's

schreibfaul1, 2021-11-19:
The example applies to the most common displays. In general, the following applies: the TP controller returns the current voltage value in mV, which it determines when the display is touched. With your special display, the voltage values ​​at the four corner points must be converted into the current pixel coordinates using a system of equations.

steevemadison, 2021-11-20:
the lcd screen i use is for a Raspberry , (generic 3.2inch ILI9341 with XPT2046) , but it's working fine with other TFT Library ,
particulary TFT_eSPI , Adafruit_ILI9341 , do you think that the current voltage from ADC Raw value is the same ?
160, 3487, 234, 3553
those value are from touch_calibrate.ino of tft_espi example

(i'm not very good at programming)

schreibfaul1, 2021-11-20:
The display is unknown to me.
I don't know if it works, but please try:
add in setup:
[image]

change in tft.cpp:
[image]

steevemadison, 2021-11-20:
wow thank you , it's working but i had to put tp rotation to 1

    tft.setRotation(3); // Use landscape format
    tp.setVersion(1);
    tp.setRotation(1);

and i did put your value in tft.cpp

compiled , upload and enjoy !!!

here's the link for the screen i used
lcdwiki.com/3.2inch_RPi_Display
there is no backlight pin on this lcd , but i put a LED , anode to GPIO17 , cathode to ground
and it does work from the LCD menu)

ESP32 MH-ET LIVE esp32 devkit wifi+ble.
Geetech VS1053 MP3 Shield with on board SD-Card.
Samsung 32GB micro-sd EVO+.
3300uf 6.3v Electrolytic Capacitor on 5V rail.
i use the same pinout config as in your main.cpp (compiled with arduino 1.8.5)

everything assemble on a breadboard for now

thank's for your time and it's a great project you did

schreibfaul1, 2021-11-20:
I am glad to hear this. Then that is also an example of the others who want to use the same display.

closed 2021-11-20 by schreibfaul1