schreibfaul1/ESP32-MiniWebRadio #59: Programm läuft, Ton startet nicht

EwanX1, 2021-12-28:
Hallo Wolle,
vorab möchte ich sagen das ich dein Projekt Top finde und diene Realisierung auch sehr schön aussieht.

Ich hatte vor ein paar Tagen das Projekt kompiliert und es funktionierte tadellos über Kopfhörer (habe noch keinen Verstärker).
Seit zwei Tagen kommt einfach kein Ton mehr und ich kann nicht herausfinden warum. Am Ende kommt dann nur noch die ganze Zeit "vs1053_info: slow stream, dropouts are possible", egal bei welchem Sender und nichts ist zu hören.

Dein Beispiel Sketch "vs1053_ext.ino" funktioniert, Sound ist zu hören.

Ich verwende ein ESP32-WROOM-32D mit VS1053B Dekoder und einem ILI9341 Touch-Display.

Nachfolgend der Log eines Senders:

`vs1053_info: Connect to new host: "edge.audio.3qsdn.com/senderkw-mp3"
vs1053_info: Connect to "edge.audio.3qsdn.com" on port 80, extension "/senderkw-mp3"
vs1053_info: Connected to server in 38 ms
vs1053_info: HTTP/1.0 200 OK
vs1053_info: server: 3qsdn edge node
vs1053_info: date: Tue, 28 Dec 2021 17:11:38 GMT
vs1053_info: content-type: audio/mpeg, format is mp3
vs1053_info: access-control-allow-origin: *
vs1053_info: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
icy-descr  : Total Lokal in Brandenburg
vs1053_info: icy-name: radioSKW
icy-name   : radioSKW
vs1053_info: icy-private:0
icy-url    : http://www.oddsock.org
vs1053_info: Switch to DATA, metaint is 16000
lastURL    : edge.audio.3qsdn.com/senderkw-mp3
vs1053_info: stream ready
vs1053_info: buffer filled in 18 ms
vs1053_info: StreamTitle='Namika - Lieblingsmensch'
vs1053_info: StreamTitle='Chris Rea - The Road To Hell'
vs1053_info: slow stream, dropouts are possible
vs1053_info: slow stream, dropouts are possible
...`

Kannst du mir eventuell Hilfestellung geben das Problem zu finden?

schreibfaul1, 2021-12-29:
"slow stream..." wird ausgelöst, wenn der Audiobuffer (fast) leer ist. Die Ursache könnte im Bereich WiFi liegen oder häufig auch an zu geringer oder schwankender Betriebsspannung.

EwanX1, 2021-12-29:
Danke für deine Antwort.
Ich hatte das ganze mehrere Tage über ein Breadboard ohne Probleme betrieben, jedoch gebe ich zu ohne einen Pufferkondensator. Wenn plötzlich die Betriebsspannung nich IO ist, ist es schon merkwürdig. Ich werde morgen mal einen Kondensator einfügen. Das Problem hat angefangen als ich Modifizierungen in die Software eingefügt hatte. Jedoch will trotz zurückspielen der originalen Software es nicht laufen.

In der Zwischenzeit habe ich im Serial-Log folgende 2 Fehlermeldungen noch gesehen:

1. CORRUPT HEAP: multi_heap.c:431 detected at 0x3ffdb40c
abort() was called at PC 0x4008cdc8 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088a98:0x3ffb7710 ...

2. /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1443 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40089a6d on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40088a98:0x3ffb1dc0 ...

Sagen dir diese Fehlermeldungen etwas? Ob diese mit dem Problem etwas zu tun haben kann ich nicht sagen.

EwanX1, 2022-01-15:
Nachträglich nochmal an alle eine Frohes neues Jahr.

Wie ich bereits geschrieben hatte is es ein "ESP32-WROOM-32D" DevKit-4 Modul. Soviel ich weiß hat dieses Modul keinen PSRAM. Nur die ESP32-WROVER Module haben einen externen PSRAM. Ich hatte es testweise mal aktiviert, jedoch hat es beim kompilieren einen Fehler gemeldet.

Im Sourcecode "vs1053_ext.cpp" wird wie folgt Speicher allokiert.
`    if(psramInit()) {
        // PSRAM found, AudioBuffer will be allocated in PSRAM
        ...
    } else {  // no PSRAM available, use ESP32 Flash Memory"
        m_buffSize = m_buffSizeRAM;
        m_buffer = (uint8_t*) **calloc**(m_buffSize, sizeof(uint8_t));
        m_buffSize = m_buffSizeRAM - m_resBuffSizeRAM;
    }
`

Da es keinen PSRAM gibt, wird im else-Zweig calloc aufgerufen.
Was bedeutet "**no PSRAM available, use ESP32 Flash Memory**"?
Wo allokiert calloc? Wird hier der Flash oder wird hier der RAM verwendet?

schreibfaul1, 2022-01-15:
PSRAM ist nicht erforderlich. Falls der AudioBuffer nicht dem PSRAM zugewiesen werden kann, wird der SRAM mit verringerter Speichrgröße benutzt. Wo der Audiobuffer beginnt kann unterschiedlich sein, m_buffer zeigt auf den Anfang.

hier der Log von meinem Gerät (vielleicht hilfs ja weiter)
````
[   349][Isetup      : Init SD card
tft_info   : TFT_CS:22 TFT_DC:21 TFT_BL:17 TFT_MOSI:23 TFT_MISO:19 TFT_SCK:18
tft_info   : init HX8347D
setup      : found SD card
setup      : Init VS1053
vs1053_info: chipID = 29891388
vs1053_info: version = 3
setup      : found VS1053
[  1496][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: mySSID
[  1519][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Wolles-FRITZBOX
WiFI_info  : Connecting WiFi...
[  3560][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  3561][I][WiFiMulti.cpp:119] run(): [WIFI] 4 networks found
[  3562][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: F0:B0:14:65:84:57 SSID: Wolles-FRITZBOX Channel: 6 (-53)
[  5782][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
WiFI_info  : WiFi connected
WiFI_info  : IP address 192.168.178.48
WiFI_info  : connected to Wolles-FRITZBOX
rtime_info : Initializing SNTP
rtime_info : Waiting for system time to be set... (1/10)
rtime_info : Sat Jan 15 22:33:59 2022
vs1053_info: Song stopped correctly after 0 msec
vs1053_info: PSRAM not found, inputBufferSize: 11903 bytes
vs1053_info: Connect to new host: "stream.laut.fm/monacensis"
vs1053_info: Connect to "stream.laut.fm" on port 80, extension "/monacensis"
vs1053_info: Connected to server in 50 ms
....
````

EwanX1, 2022-04-21:
Sorry das ich mich nicht so lange gemeldet hatte. Ich war etwas länger auf Dienstreise im Ausland.
Ich werde mich aber demnächst nochmal des Problems annehmen und es nochmal versuchen.

closed 2022-08-13 by stale[bot]

fv1895, 2023-11-08:
Hallo zusammen,
ich setze mich mal auf diesen Bericht, da er auch deutsch ist.
Als erstes muss ich sagen, dass ich das Programm/Projekt wirklich richtig gut finde. Ich hatte alle Komponenten schon auf einem Steckboard am laufen und war begeistert. Als ich dann die Komponenten auf  einer Lochrasterplatine verbunden habe, funktionierte weder Ton (PCM5102) noch Bild (ILI9488). Inzwischen habe ich schon alle Komponenten ausgetauscht, leider ohne Erfolg.  Ich versuche jetzt schon seit gefühlt 100 Stunden alles ans laufen zu bekommen, kann aber keinen Fehler finden. 
Vielleicht finde ich ja auf diesem Wege Hilfe.
[image]
[image]

AVRsteffen, 2023-11-08:
Ich denke es liegt an dem sehr langen Flachbandkabel und dem ESP N16R8.
Nimm einen ESP32 WROVER N4R8

Arne65, 2023-11-08:
Hallo Steffen 

wie sieht den deine serielle Monitorausgabe aus Bootet der ESP korrekt ?

Trenne Display und den PCM vom ESP . Der ESP sollte erstmal korrekt Booten das geht auch ohne Display und PCM5102.

Arne65, 2023-11-08:
Dann kann es eigentlich nur an deiner Verkabelung liegen 

da weder Display noch Ton funktioniert .Eine blöde Frage  bekommen beide überhaupt Spannung ? Mal Am Display etc . .gemessen ?

valera39, 2023-11-08:
The voltage to the ESP32 must be applied directly to the +5v pin, not through the USB connector

fv1895, 2023-11-08:
> Wenn du an dem PCM5102 Board den Anschluss XSMT mit dem Finger berührst und ein Knacken hörst, dann muss XSMT extern auf HIGH gelegt werden. [image] Normalerweise gibt es dafür auf dem Board einen Pullup Widerstand. Aber ich habe zwei Exemplare, wo dieser fehlt.

Das scheint leider nicht das Problem zu sein. Könntest Du mir bitte die Gpios zu den Anschlüssen geben. Ich bin mir inzwischen überhaupt nicht mehr sicher ob ich die richtigen genommen habe.

schreibfaul1, 2023-11-08:
Die Pins sind frei wählbar. Im Original sind das I2S_LRC 1,  I2S_BCLK 3 und I2S_DOUT 9.
Du kannst in common.h andere Pins wählen. Mit einem Oszi oder Logik Analysator kannst du die Impulse sehen.

fv1895, 2023-11-09:
So, ich habe jetzt alle Anschlüsse überprüft und kann keine Fehler feststellen. Kann es vielleicht sein, daß die Kabelbrücken (Klingeldraht) zu dick sind?

Arne65, 2023-11-09:
Hallo fv1985

Du hast sicher von jedem einzelnen PIN vom ESP bis zum Display / PCM gemessen .Und da hast du überall Durchgang ? Am Display und PCM liegt auch die Spannung an ? Besser wäre noch mit einem Oszi zu messen .

Die GPIO´s stimmen sicher auch .

Bei meinen PCM Modulen ist eine LED verbaut als Betriebsanzeige  denke das wird bei Dir auch so sein die leuchtet hoffentlich .
(Spannung entweder 3,3 oder 5 V  liegen an beiden an / meine Displays egal welches ich benutzt habe und auch meine DAC`s funktionieren alle auch problemlos mit 3,3 V )

Deine Klingeldrähte sind 0,6 mm was ich auf dem Foto sehe insofern sind die nicht zu Dick  und der Querschnitt der Leitungen hat nur bedingt was mit der Funktion zu tun . (Strom etc.)

Da dein ESP Bootet kann es nur eine Spannung  oder Kontakt Problem sein . Oder du hast beides (Display und PCM ) zerschossen .

Da beides nicht funktioniert liegt das fast nahe . Da Display und PCM unabhängig funktionieren wenn alles korrekt ist . Bedeutet selbst wenn das Display Tot ist und der PCM funktioniert kommt das Mucke raus oder eben andersrum .

Miss nochmal wirklich alles nach irgendwo wirst du einen Bock drin haben .

fv1895, 2023-11-09:
Danke für die Hinweise, einen Oszi habe ich leider nicht, nur ein Multimeter. Zerschossen habe ich hoffentlich nichts, ich war da aber sehr vorsichtig. Ich bestelle mir am besten mal ein neues PCM und versuche es dann nochmal. Hat jemand ein Tipp für mich welches PCM ich am besten nehme?

Arne65, 2023-11-09:
Leuchtet denn die LED auf dem DAC ? 

Ich habe die genommen wenn es schnell gehen soll 

https://www.amazon.de/dp/B07XB4BYXC?psc=1&ref=ppx_yo2ov_dt_b_product_details

ansonsten habe ich die bei Ali Express bestellt ist deutlich preiswerter Dauert halt nur etwas länger

https://de.aliexpress.com/item/1005005393398013.html?spm=a2g0o.order_list.order_list_main.22.5bd85c5fLYwtiW&gatewayAdapt=glo2deu

schreibfaul1, 2023-11-09:
Der DAC lässt sich nicht so schnell zerstören, außer du hast die Betriebsspannung verpolt.
Die Kabellänge/dicke spielt keine große Rolle, die Frequenzen sind nicht so hoch wie beim Display.

Du kannst den Aufbau so lassen wie er jetzt ist und lädtst einfach mal das Beispiel vom [Readme](https://github.com/schreibfaul1/ESP32-audioI2S) herunter. Das ist die selbe Bibliothek, die MWR benutzt, aber die Fehlersuche wird einfacher.

valera39, 2023-11-09:
Still, check the voltage on the DAC and display modules

steve6375, 2023-11-09:
Presumably you are not trying to power it using the external power supply AND the USB port at the same time????
You either power it from the USB cable OR unplug the USB cable and use the external PSU?

valera39, 2023-11-09:
[steve6375](https://github.com/steve6375), I think it only uses USB power

fv1895, 2023-11-10:
I User only the external PSU. Maybe I have used both for testing with the terminal.

steve6375, 2023-11-10:
Have you just tried it just powered from the USB cable? Mine works fine like that. I use the 5V pins on the esp32 to power the DAC and TFT (5v pin is actually about 4.75V as it goes via a diode on the esp32 board).

fv1895, 2023-11-11:
> Der DAC lässt sich nicht so schnell zerstören, außer du hast die Betriebsspannung verpolt. Die Kabellänge/dicke spielt keine große Rolle, die Frequenzen sind nicht so hoch wie beim Display.
>
> Du [...]

Erstmal an alle vielen Dank für eure Geduld! 
Jetzt bekomme ich aber eine Fehlermeldung wie auf den Screenshot zu sehen ist.
[image]

schreibfaul1, 2023-11-11:
Das ist viel zu kompliziert. Lösche alles, was inVS Code in main.cpp steht und übertrage das:
````c++
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Audio audio1;

String ssid =     "++++";
String password = "++++";

void setup() {
    Serial.begin(115200);

    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid.c_str(), password.c_str());
    while (WiFi.status() != WL_CONNECTED) delay(1500);
    audio1.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio1.setVolume(21); // default 0...21
//
//  *** radio streams ***
    audio1.connecttohost("http://stream.antennethueringen.de/live/aac-64/stream.antennethueringen.de/"); // aac
//  audio.connecttohost("http://mcrscast.mcr.iol.pt/cidadefm");                                         // mp3
//  audio.connecttohost("http://www.wdr.de/wdrlive/media/einslive.m3u");                                // m3u
//  audio.connecttohost("https://stream.srg-ssr.ch/rsp/aacp_48.asx");                                   // asx
}

void loop()
{
    audio1.loop();
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
````
und dann noch die WiFi Zugangsdaten ergänzen.

fv1895, 2023-11-11:
Anscheinen stehe ich total auf dem Schlauch! Ich habe den Code in Arduino IDE kopiert, bekomme aber beim kompilieren immer die Fehlermeldung: "Compilation error: 'Audio' does not name a type"

In Visual Studio kann ich kein Board auswählen und auch nichts kompilieren. Bisher habe ich auch noch nie mit Visual Studio gearbeitet sondern auschließlich mit Arduino IDE

fv1895, 2023-11-11:
Ich habe das jetzt so in die C:\Users\bierh\ESP32-audioI2S\examples\DLNA\src\main.cpp eingefügt, bekomme aber beim debuggen folgende Fehlermeldung:
This chip is ESP32-S3 not ESP32. Wrong --chip argument?
Das kann ich auch nachvollziehen, habe allerdings keine Ahnung wo ich das ändern muss.

Arne65, 2023-11-11:
Ich denke du hast als Target noch esp 32 das musst du ändern unten links steht Esp32 klicken und oben als Target Esp32s3 wählen

fv1895, 2023-11-11:
So, ich hatte noch einen ESP32 zu Hause. Auch hier mit dem Programm keinen Ton. Dann wird das PCM tatsächlich kaputt zu sein.
Melde mich dann wieder wenn ich das neue PCM habe.