schreibfaul1/ESP32-MiniWebRadio #60: SD card error

Beppi4U, 2022-01-03:
Hallo!
Nachdem mein letzter Versuch lange her ist und nicht von Erfolg gekrönt war habe ich jetzt einen anderen ESP32S zum Aufbau genommen! Verkabelt.....eigentlich wie auf Deiner Zeichnung.....tut aber trotzdem nicht!
Ich habe mal den Monitor Output angehängt! 
TFT springt kurz an, zeigt nichts und geht wieder aus!
Ist in dem Pinout  Gpio2/XS -> VS1053/XCS gemeint ? So habe ich es angeschlossen.
Da auf meinem TFT/Display und SD Karte getrennt sind habe ich alle, SDkarte/TFT+TFTtouch/VS1053 an Gpio18/SCK Clock angeschlossen.
SD CS liegt auf Gpio P5 
Liegt da eventuell irgendwo der Fehler?
IR habe ich vorerst mal ausgelassen. 
Vielen Dank schon mal für Hilfe! ;-)
Andreas

[image]

[image]

[image]

schreibfaul1, 2022-01-04:
Wie es aussieht ist SPI nicht komplett, 
GPIO23 muss verbunden sein mit VS1053_MOSI und SD_MOSI und  T_DIN und SDI
GPIO19 muss an VS1053_MISO und SD_MISO und T_DOUT und SDO
GPIO18 muss an VS1053_SCK und SD_SCK und T_CLK und SCK

Beppi4U, 2022-01-04:
O.K.! Ich habe die fehlende Verbindung SCK hergestellt.
Bekomme jetzt schon mal eine anders lautende Fehlermeldung.....Veränderung ist gut!
Sollte ich das richtig interpretieren - findet der ESP die stations.csv nicht auf der Karte und hat auch keine Berechtigung eine zu generieren.
Wenn ich aber auf die Karte schau finde ich Sie....!? Sollte die CSV in einen Unterordner /sd/stations.csv?
Die Karte ist eine Steinalte mit 512MB, FAT32 Formatiert und hat keinen Schreibschutz!
Gruß 
Andreas

[image]

[image]

schreibfaul1, 2022-01-04:
Bei Deinem Display ist der SPI Bus von TFT, TP und SD getrennt herausgeführt. Ich habe auch so ein Display, bei dem habe ich alle Leitungen mit dünnem Kupferlackdraht verlötet, alle SCKs miteinander, alle MOSIs und alle MISOs. Das hat zwei Vorteile, keine Kontaktschwierigkeiten und kürzere Drahtlängen. 80 MHz sind mit einer freien Verdrahtung nicht erreichbar, setze SD.begin(SD_CS, SPI, 8000000); lieber auf 20MHz
https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9341-HX8347D/blob/master/additional%20info/TFT_ILI9341_wirering.jpg

Beppi4U, 2022-01-05:
O.K.! Here we go! 
Nach dem 10 mal neu verkabeln läuft jetzt das WEBRadio!
....das Display bleibt aber nur hell und flackert im MP3 Modus(nur im MP3 Modus)! 
MP3 stoppt nach einem Song und geht in den WebRadio Modus über.
Während Mp3 werden CRC Fehler dauerhaft ausgegeben( [652163][W][sd_diskio.cpp:186] sdCommand(): token error [17] 0x47 )

Hier mal der restliche Serial Output....
Vielleicht bekommen wir das Display auch noch zum laufen.... ;-)
Gruß 
Andi

Snip......
--- Miniterm on COM10  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
[     0][E][WiFiClient.cpp:318] setOption(): fail on fd -1, errno: y� "Bad file number"
[     0][E][WiFiClient.cpp:297] setSocketOption(): 1006 : 9
[    97][Isetup      : Init SD card
[   101][W][sd_diskio.cpp:186] sdCommand(): token error [0] 0x47       
[   101][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[   104][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
tft_info   : TFT_CS:22 TFT_DC:21 TFT_BL:17 TFT_MOSI:23 TFT_MISO:19 TFT_SCK:18
tft_info   : init ILI9341
[   337][W][sd_diskio.cpp:180] sdCommand(): crc error
setup      : found SD card
[   467][W][sd_diskio.cpp:174] sdCommand(): no token received
[   626][W][sd_diskio.cpp:180] sdCommand(): crc error
[   738][W][sd_diskio.cpp:180] sdCommand(): crc error
[   848][W][sd_diskio.cpp:174] sdCommand(): no token received
[   959][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1069][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1188][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1297][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1415][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1531][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1643][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1751][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1872][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1984][W][sd_diskio.cpp:180] sdCommand(): crc error
[  2092][W][sd_diskio.cpp:180] sdCommand(): crc error
[  2202][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2312][W][sd_diskio.cpp:180] sdCommand(): crc error
[  2421][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2531][W][sd_diskio.cpp:180] sdCommand(): crc error
[  2641][W][sd_diskio.cpp:180] sdCommand(): crc error
[  2750][W][sd_diskio.cpp:180] sdCommand(): crc error
[  2858][W][sd_diskio.cpp:174] sdCommand(): no token received
[  2969][W][sd_diskio.cpp:180] sdCommand(): crc error
[  3080][W][sd_diskio.cpp:186] sdCommand(): token error [17] 0x47
setup      : Init VS1053
vs1053_info: chipID = 378472233
vs1053_info: version = 3
setup      : found VS1053
[  3758][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: WLAN-DLRPMN
[  3759][W][sd_diskio.cpp:180] sdCommand(): crc error
[  3884][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01
[  3885][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02
[  3889][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03
[  3897][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
[  3905][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04
WiFI_info  : Connecting WiFi...
[... 27 lines cut ...]
vs1053_info: Connected to server in 35 ms
[ 21042][W][sd_diskio.cpp:180] sdCommand(): crc error
[ 21197][W][sd_diskio.cpp:180] sdCommand(): crc error
[ 21311][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 21419][W][sd_diskio.cpp:186] sdCommand(): token error [17] 0x47
vs1053_info: HTTP/1.0 200 OK
vs1053_info: icy-notice1:<BR>This stream requires <a href="http://www.winamp.com">Winamp</a><BR>
vs1053_info: icy-notice2:Shoutcast DNAS/posix(linux x64) v2.6.0.753<BR>
vs1053_info: access-control-allow-origin:*
vs1053_info: icy-name: Best Oldies
icy-name   : Best Oldies
vs1053_info: icy-sr:44100
icy-url    : http://www.shoutcast.com
vs1053_info: content-type:audio/mpeg, format is mp3
vs1053_info: x-clacks-overhead:GNU Terry Pratchett
vs1053_info: Switch to DATA, metaint is 16384
[ 21555][W][sd_diskio.cpp:180] sdCommand(): crc error
[ 21709][W][sd_diskio.cpp:180] sdCommand(): crc error
[ 21824][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 21932][W][sd_diskio.cpp:186] sdCommand(): token error [17] 0x47
lastURL    : c3.auracast.net:8010/stream
vs1053_info: stream ready
vs1053_info: buffer filled in 13 ms
vs1053_info: StreamTitle='Elton John - Goodbye Yellow Brick Road'
......Snip off

schreibfaul1, 2022-01-05:
Für crc Fehler sind ganz sicher Kontaktprobleme auf den SPI Leitungen verantwortlich. Mit Steckverbindern lässt sich das kaum verhindern. Am besten ist der Versuchsaufbau auf einer Lochrasterplatte. Das hat außer der sicheren Kontaktgabe auch den Vorteil der kürzeren Leitungswege. Für das Display gilt dasgleiche. Probier mal bitte eine niedrigere Frequenz, 20 oder 10MHz, tft.setFrequency(20000000)

Beppi4U, 2022-01-05:
Da ich die Anweisung "tft.setFrequency(20000000)" so nirgendwo finden konnte habe ich sie hinter tft.begin gesetzt! Ich hoffe das war so richtig.
Leider hat die Änderung nichts gebracht...das Display bleibt "nur hell".
Ja das mit den Kontakten kann ich wohl nachvollziehen. Meine Strippen sind aber kaum länger als 10cm mit Dupont Steckern. Die Y-Kabel habe ich alle nochmal durchgeklingelt weil schlechte Lötverbindungen ausschliessen wollte...sind alle gut. 
hmm.!? und nu?

schreibfaul1, 2022-01-05:
haste TFT tft(0);   für ILI9341?

Beppi4U, 2022-01-07:
Ja daran habe ich gedacht! 

[image]

.....bleibt ja immer die Möglichkeit eines defekten Displays! 
Ich schau mal ob es mit einem einfachen Sketch zur Anzeige überreden kann!

AVRsteffen, 2022-01-26:
Diese Probleme hatte ich anfangs auch. Falls du eine der neusten Arduino ESP32 Board Bibliothek nehmen solltest (2.0.2) könnte es daran liegen. Ich habe die besagte 1.0.6 genommen und dann hat es endlich funktioniert. Wolle hatte in irgend einem Issue was von veränderten Funktionsnamen in der SD Bibliothek der neueren Board Bibos ab 2.0.0 alpha geschrieben. Glaub was mit File.name <-> File.path oder so..

Also versuche es mal mit der ESP32 Board Bibo 1.0.6 zu kompilieren

Beppi4U, 2022-01-31:
Hi! Sorry für die späte Rückmeldung ! 
Danke für den Tipp! Ich bin etwas überfordert mit der VS Code und PlatformIO Oberfläche! Wo kann ich so was umstellen bzw. einstellen??