schreibfaul1/ESP32-MiniWebRadio #61: Networks.csv

SycoJ, 2022-01-06:
Hallo Wolle,

Kannst du mir mal fix einen Tipp geben?

Ich habe ein weiteres Netzwerk hinzugefügt.

Das wird jedoch nicht gefunden. Ich vermute meine Schreibweise bei der SSID ist falsch.

Bei den Fritzbox Netzwerken lautet beispielsweise der Standard:

FRITZ!Box (Leerzeichen) 7590 Leerzeichen FZ.

Müssen das auch Tabs innerhalb der Zelle sein? Oder kann es ein normales Leerzeichen oder ein anderes Zeichen sein?

Bei mir funktioniert beides nicht. 

In der Miniwebradio.ino habe ich aber mein Netzwerk hinterlegt, was auch funktioniert. 

Darf es nur entweder *.ino oder Networks.csv sein?

Danke im Voraus.

schreibfaul1, 2022-01-07:
Hallo @SycoJ,
wenn es nur ein Netzwerk gibt reicht der Eintrag in der .ino
Einträge in der Networks.csv sind dann sinnvoll, wenn du dich in verschiedenen WLANs einloggen möchtest, also bspw. an unterschiedlichen Standorten. Dazu wird intern WiFiMulti benutzt. Ich glaube Leerzeichen innerhalb der SSID sind unzulässig. Unter "WLAN/Funknetz" kannst du die SSID ändern.
Hab selbst eine Fritzbox 7490 und als Zweit-WLAN den Gastzugang eingerichtet.

vG
Wolle

SycoJ, 2022-01-07:
Ich habe auch mehrere Netzwerke. 

Kannst du mir mal ein Screenshot oder Auszug von deinem Eintrag bezüglich der Fritzbox schicken?

Irgendwas gebe ich falsch ein und möchte gern mal deinen Eintrag mit meinem in meiner Networks.csv vergleichen. 

Bzw. setzt du ein Tab oder ein anderes Zeichen zwischen Fritzbox und 7490 oder hast du deine SSID in deiner Fritzbox umbenannt?

Danke im Voraus

schreibfaul1, 2022-01-07:
Habs eben probiert, Leerzeichen sind erlaubt (war Quatsch was ich geschrieben hatte)
Der Trenner ist TAB,  in einer Zeile steht SSID {TAB} PW {TAB} irgendwas {LF}
[networks.csv](https://github.com/schreibfaul1/ESP32-MiniWebRadio/files/7828800/networks.csv)

SycoJ, 2022-01-08:
Wie bearbeitet du die Datei? Ich hab es über Excel und den Standard Editor von Windows 10 versucht. Er connected nicht.

Das einzige was zuverlässig funktioniert ist die Eingabe in der Miniwebradio.ino.

Apropos, muss ich in der Miniwebradio.ino noch etwas ändern bzw. anpassen? 

Entschuldige bitte die Umstände aber ich weiß wirklich  nicht woran es liegen kann. :/

SycoJ, 2022-01-13:
So. Es funktioniert nur, wenn ich alles in die Networks.csv schreibe und nichts in der Miniwebradio.ino. klingt komisch ist aber so ^^

SycoJ, 2022-01-13:
Wolle entschuldige bitte. Ich hatte PSRAM aus und die MHZ noch auf 40 statt 80. :/ Es funktioniert einwandfrei.
Wie ich das nur übersehen konnte...

Kann hier geschlossen werden.

PS: Kannst ja mein Beitrag als "known issue" aufnehmen ^^ Oder erstelle doch eine FAQ ^^

closed 2022-01-13 by SycoJ