schreibfaul1/ESP32-MiniWebRadio #65: MMS Card for content und SD extern for mp3 files

AVRsteffen, 2022-02-07:
Hallo

Erstmal herzlichen Dank für dieses tolle Projekt Wolle. Wie ich finde, eines der Besten! 
Besonders toll die TFT Anbindung mit den Fonts und Bildchen. 

Bezugnehmend auf den schon geschlossenen Issue #46 hab ich es geschafft, eine SD Karte im "1-bit SDMMC mode" an GPIO2, GPIO14 und GPIO15 zum laufen gebracht. Wie gesagt, das Problem war der GPIO2. Ich nutze das DOT ESP32 DevKit V1. Um den GPIO2 für im SD_MMC mode nutzen zu können, muss man die LED und den 10k gegen Masse entfernen und extern 10k nach Masse zum Programmieren und 10k gegen 3,3V zum Booten schalten. Ich hab das über eine Steckbrücke gemacht. Aber so funktioniert es erstmal.
Eingebunden hab ich die SD_MMC.h
`#include <SD_MMC.h> `

Dann hab ich eine neues Flag unter den "//global variables" hinzugefügt:
`boolean  f_MMC_okay=false;      // true if SD-MMC card in place and readable`

Dann alle "f_SD_okay" Flags durch "f_MMC_okay" ersetzt - Außer die der MP3 File Funktionen
Das selbe auch mit allen Funktionen wo "SD" vorkommt wird durch "SD_MMC" ersetzt - Außer die der MP3 File Funktionen.
Beispiel:
(vorher)
```
    if(f_SD_okay){`
        if(tft.drawJpgFile(SD, str2.c_str(), 0, _yLogo)==false){ // filename mostly given from _stationname exist?
            tft.drawJpgFile(SD, "/logo/unknown.jpg", 0, _yLogo);  // if no draw unknown
        }
```
(ersetzt)
```
    if(f_MMC_okay){
        if(tft.drawJpgFile(SD_MMC, str2.c_str(), 0, _yLogo)==false){ // filename mostly given from _stationname exist?
            tft.drawJpgFile(SD_MMC, "/logo/unknown.jpg", 0, _yLogo);  // if no draw unknown
        }
```

Und hier der Setup:
```
//**************************************************************************************************
//                                           S E T U P                                             *
//**************************************************************************************************
void setup(){
    #if ESP_ARDUINO_VERSION_MAJOR >= 2
        log_i("Arduino Version >=2.0.0");
    #else
        log_i("Arduino Version < 2.0.0");
    #endif
    // first set all components inactive
    Serial.begin(115200); // For debug
    // Initialize the SD card
    Serial.println("setup      : Init MMC card");
    if (!SD_MMC.begin("/sdcard", true)){
        Serial.println("Failed to mount SD card");
        return;
    }
    pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);
    pinMode(TFT_CS, OUTPUT);     digitalWrite(TFT_CS, HIGH);
    pinMode(TFT_DC, OUTPUT);     digitalWrite(TFT_DC, HIGH);
    pinMode(TP_CS, OUTPUT);      digitalWrite(TP_CS, HIGH);
    pinMode(VS1053_CS, OUTPUT);  digitalWrite(VS1053_CS, HIGH);
    pinMode(ENC_CLK, INPUT_PULLUP);                    // Input for encoder
    pinMode(ENC_DT, INPUT_PULLUP);                     // Input for encoder
    pinMode(ENC_SW, INPUT_PULLUP);                     // Input for encoder
    
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    SD.end();       // to recognize SD after reset correctly
    Serial.println("setup      : Init SD card");
    SD.begin(SD_CS);
    delay(100); // wait while SD is ready
    tft.begin(TFT_CS, TFT_DC, SPI_MOSI, SPI_MISO, SPI_SCK, TFT_BL);    // Init TFT interface  
    SD.begin(SD_CS, SPI, 80000000);  // fast SDcard set 80000000, must have short SPI-wires
    tft.setRotation(1); // Use landscape format
    tp.setRotation(3);
    pref.begin("MiniWebRadio", false);  // instance of preferences for defaults (tone, volume ...)
    stations.begin("Stations", false);  // instance of preferences for stations (name, url ...)
    setTFTbrightness(pref.getUInt("brightness"));
    // Check for an SD card
    f_MMC_okay = (SD_MMC.cardType() != CARD_NONE);
    if (!f_MMC_okay){
        Serial.println("setup       : MMC card not found");
        while(1){};                     // endless loop, MiniWebRadio does not work without SD
    }
    else Serial.println("setup      : found MMC card");
    f_SD_okay=(SD.cardType() != CARD_NONE); // See if known card
    if(!f_SD_okay){
        Serial.println("setup       : SD card not found");
     //   while(1){};                     // endless loop, MiniWebRadio does not work without SD
    }
    else Serial.println("setup      : found SD card");
    File file=SD_MMC.open("/stations.csv");
    if(!file){
        Serial.println("setup       : stations.csv not found");
        while(1){};                     // endless loop, MiniWebRadio does not work without stations.csv
    }
    int size= file.size();
    file.close();
    if(stations.getLong("stations.size") != size) defaultsettings();  // first init
    tft.drawJpgFile(SD_MMC, "/MiniWebRadio.jpg", 0, 0); //Welcomescreen
    Serial.println("setup      : Init VS1053");
    vs1053.begin(); // Initialize VS1053 player
```
Mun wollte ich eine externe microSD nur für mp3 Audio Datein vorsehen und die SD-MMC beinhaltet alle Logos, Button-Bilder, Netzwerkliste, Stationsliste und was da sonst noch an Daten drauf sind. Das funktioniert auch alles soweit ganz gut.

**Bis auf die Webseite (Websocket)!**
Da wird immer noch die andere SD Karte benutzt um die Listen und Bilder zu laden. Wenn ich auch in der **index.h** alle "SD" durch "SD_MMC" ersetze, werden keine Bilder und Listen mehr geladen. 

[image]

Was habe ich übersehen?
Was mache ich verkehrt?

Benötige **Hilfe**. Wie wird auch von der Webseite die Bildchen und Listen von der MMC Karte benutzt?

AVRsteffen, 2022-02-07:
Ach ja, das TFT funktioniert schon mit der MMC Karte. Das kann ich an den verschiedenen Logos (eckig bei SD - abgerundet bei MMC) sehen. Mein geändertes Favicon.ico wird bei der Webseite übrigens schon von der MMC Karte geholt.

schreibfaul1, 2022-02-07:
SD_MMC läuft aus meiner Sicht stabiler als SD über SPI und die Schreib/Lesegeschwindigkeit deutlich höher. Die GPIOs können nicht frei gewählt werden, GPIO2 ist hier der Problemkandidat. Wie du geschrieben hast ist ein PullUp Widerstand beim Bootvorgang Widerstand schädlich, wird aber im Betrieb benötigt. Viele SD Card Adapter bringen leider diesen PullUp mit. Danke für Deinen interessanten Bericht und viel Freude mit diesem Projekt. 
Und nun zur Frage. Wenn eine Datei vom Webserver geholt werden soll, wird auch die Quelle mit angegeben. Suche im Quelltext nach "webSrv.streamfile" und tausche SD gegen SD_MMC.
Und tausche überall wo connecttoSD(path); steht gegen ConnecttoFS(SD_MMC, path). Hoffe das hilft.

AVRsteffen, 2022-02-07:
Hallo Wolle,

Vielen Dank für den Hinweis. In welchem Quelltext soll ich da schauen? 
webserv.cpp?
miniwebradio.ino?
oder
index.h?

Wenn es klappt, wäre das echt genial. Denn dann steht der SPI auch für andere Devices bereit. Ich finde ja auch die USB File Anbindung mittels CH376 interessant.

Hier mal noch die Schaltung zur MMC Anbindung
[image]

Gruß Steffen

schreibfaul1, 2022-02-07:
nur die *.ino
[image]
vs1053.connecttoFS(SD_MMC, param);
webSrv.streamfile(SD_MMC, ....
überall wo das auftritt. 

Du brauchst keinen Schalter an DAT0, es reicht wenn der 10K Widerstand gegen +3V3 ausgelötet ist.
und im setup() setzt du den internen pullUp
    pinMode(2, INPUT_PULLUP);
    if(!SD_MMC.begin("/sdcard", true)){
        log_e("SD Card Mount Failed");
    }

AVRsteffen, 2022-02-07:
Interessant..
> Du brauchst keinen Schalter an DAT0, es reicht wenn der 10K Widerstand gegen +3V3 ausgelötet ist.
> und im setup() setzt du den internen pullUp
```
pinMode(2, INPUT_PULLUP);
if(!SD_MMC.begin("/sdcard", true)){
log_e("SD Card Mount Failed");
}
```
Das werde ich noch testen. Ich habe nur bemerkt, dass ein Pullup zum Betrieb der MMC gebraucht wird. An GPIO2 ist normalerweise ein 10k Pulldown eingelötet. Ich hab gelesen, dass der interne Pullup so ca. 30k..50k haben soll. (Normal)
Deswegen war ich am zweifeln, dass der interne Pullup gegen externen 10k Pulldown ankommt. Aber schöner wärs natürlich. Wenn nicht, dann tausch ich den 10k Pulldown gegen einen 47k oder so.

Außerdem hab ich die Pullups an der CLK und  CMD Leitung vergessen.

Nun zum eigentlichen Problem:
`    if(cmd=="favicon.ico")  {webSrv.streamfile(SD_MMC, "/favicon.ico"); return;}`
-> hatte ich vorher schon geändert. Das hatte ja auch funktioniert.

Alles was mit **mp3** zu tun hat, will ich ja über die SPI SD Karte laufen lassen. Deswegen hab ich dies auch noch nicht geändert.

Mir ist aber folgendes aufgefallen:
[image]

Und ich war schon die ganze Zeit auf der richtigen Fährte. Hier ist der Fehler
[image]

Nun geht alles - Bähm!
Danke für die Erleuchtung Wolle 💯 Punkte 👍

schreibfaul1, 2022-02-18:
Der PullUp Widerstand an GPIO2 wird häufig eingesetzt und ist bei SPI Betrieb auch sinnvoll. Hier ein Zitat aus dem ESPUINO Forum 
____
SD_MMC
PRO:

Erheblich schneller als SPI
Benötigt nur 3 GPIOs
CON:

GPIOs (15, 14, 2) sind fest und können nicht geändert werden.
Einer der GPIOs ist 2. Wird dieser GPIO beim Start des ESP32 auf HIGH gezogen, so kriegt man den ESP32 nicht in den Flash-Modus. Das bedeutet, dass man SD-Board ein etwaiger PullUp-Widerstand entfernt werden muss. Siehe z.B. [hier 17](https://github.com/biologist79/ESPuino/blob/master/pictures/Pullup-removal.jpg) bei [diesem SD-Board 8](https://de.aliexpress.com/item/32886674838.html).
Fazit: SD_MMC lohnt sich auf jeden Fall. Hier noch ein paar Durchsatzraten bei einem Dateitransfer von 70,7 MiB.

SD_MMC:
FTP (up): 3:57min => 298 kiB/s
FTP (down): 3:45 => 314 kiB/s
HTTP: 3:10min => 362 kiB/s

SPI:
FTP (up): 6:38 => 178 kiB/s
FTP (down): 5:38 => 208 kiB/s
HTTP: 6:24min => 184 kiB/s_ 
____
Es ist häufig nicht einfach diesen Widerstand auf den Boards zu finden. Es freut mich, dass es jetzt funktioniert.