schreibfaul1/ESP32-MiniWebRadio #66: ILI9486 ILI9488

valera39, 2022-02-11:
Hello! Is it possible to add display support on ILI9486 and ILI9488, for larger screens 480x320px?

schreibfaul1, 2022-02-12:
It's possible. There will probably be a version with ILI9486 in March. In order to avoid dropouts when writing on the display, it is necessary to run the VS1053 in a separate task. So that there are no collisions, there must be a second SPI. Also, it is better if the SD card is switched as SD_MMC. This already works with the small display. There is a trial version as a separate branch called MiniWebRadio-V2

clear-sky, 2022-03-01:
Hi
as far as i know esp32 has two cores 
edzelf uses each core for different tasks
please see here : https://github.com/Edzelf/ESP32-Radio/blob/master/src/main.cpp
line 3511 and after

clear-sky, 2022-03-03:
thank you very much! for ili9488
you are very kind

schreibfaul1, 2022-03-04:
I've been experimenting with this lately. My display is a red board `3.4'' TFT SPI 480x320 v1.0`. Either the TFT or the TP worked, but not both together. TFT_MISO is not a tristate pin and blocks the SPI. If TFT_MISO remains unconnected, it works

https://user-images.githubusercontent.com/26044260/156740696-1129560c-19d0-4ae3-9ffa-adaa7791eee8.mp4

.

closed 2022-08-13 by stale[bot]