schreibfaul1/ESP32-MiniWebRadio #67: MP3 Play/Pause/Play möglich?

AVRsteffen, 2022-02-17:
Wenn ein MP3 File abgespielt wird und es kommt zur Zeitansage, wird das MP3 File gestoppt - die Zeit angesagt - und danach der letzte Radio Stream gestartet.

Gibt es irgendwelche Funktionen/Methoden in VS1053 die es möglich machen, sich die Stelle des MP3 Files zu merken und später wieder zu laden? Um an genau der gleichen Stelle das MP3 File fortzuseten? Also eine Play/Pause/Play Funktion?

Oder kennt irgend jemand Code zum VS1053 wo sowas umgesetzt wurde?

LG Steffen

AVRsteffen, 2022-02-18:
Hallo Wolfgang,
`if(_time_s.endsWith("59:51") && _state == RADIO) {`
Mit dieser Änderung könnte ich erstmal leben ^^

Zum Thema MP3-File und Play/Pause/Play:
schreibfaul1 schrieb:
> Der Wiedereinstieg an der letzten Dateiposition ist etwas schwierig, da es nicht ausreicht den Dateizeiger zu speichern. Es
> müsste noch eine Korrektur mit den verbleibenen Daten im Audiobuffer erfolgen.

Da ich noch nicht so richtig weiß wie der VS1053 funktioniert, bin ich gerade dabei darüber etwas mehr zu erfahren. Und dabei hab ich auch gelesen, dass das Auffüllen des Audiobuffers mit '0'en ebenfalls zu Audioartifakten bei der decodierung der mp3 kommt. Man müsste also ein genau am Anfang des Frames starten. Dabei soll dies auch bei ogg und wma Datein funktionieren wenn man dafür zusätzlich den File Header neu überträgt.

MiniWebRadio-V2:
schreibfaul1 schrieb:
> Es gibt dann zwei SPI Instanzen

Super! Das war ja auch meine Idee mal gewesen mit 2 Instanzen zu arbeiten. Ist allerdings bei den Libraries nicht vorgesehen. So muss man extra Klimmzüge machen.. Und dies hast du getan? -> Ich werde das auf alle Fälle mal ausprobieren. Zumal gestern auch mein 3,5inch 480x320 TFT mit ILI9486 gekommen ist.

Meine Version mit der MMS als Datenspeicher und einer SD als externe wechselbare Audio Karte funktioniert auch schonmal. Allerdings musste ich die SPI-TFT Frequenz auf 20Mhz herabsetzen. Denn mit 40Mhz lief dann nicht mehr stabiel und es kam immer wieder zum Totalausfall des TFT.

Ich finde die Trennung zwischen Datenspeicher zum Betrieb des Mini Web Radios und einem externen Audio Mediums äußerst interessant. Zumal man da warscheinlich auch den CH376 für USB Massenspeicher per SPI einbinden kann.

[image]
 
oder mit SD Slot

[image]

Was ein tolles Projekt!

LG Steffen

AVRsteffen, 2022-02-19:
Hallo Wolle,

> Ich arbeite gerade an einer neuen Version, die mir der gleichen HW arbeitet aber dennoch andere Pins benutzt. Es gibt dann zwei SPI Instanzen, davon eine exklusiv für Audio. Dadurch werden die Tonaussetzer beim Grafikaufbau oder beim Laden der Webseite vermieden. Und die SD Karte läuft dann als SD_MMC, was einiges an Geschwindigkeit und Stabilität bringt. Das kannst du schon jetzt testen, einfach das Git Repository mit PlatformIO klonen und den Branch MiniWebRadio-V2 auswählen.

Testen werde ich es erst wenn ich meine nächsten ESP32 Kits habe. Möchte dazu ungern mein jetziges Setup auseinander reißen. Hab da auch noch einen Encoder mit eingebunden um erstmal die Lautstärke/Mute auch darüber zu regeln.  Das funktioniert auch schon.

Ich hab mir mal die Quellen zur V2 angesehen. Wenn ich das richtig verstehe läuft der HSPI für den VS1053 dann über einen Task? Um dann doch noch eine externe SD an den gleichen SPI Bus (HSPI) anzubinden, muss man dafür dann auch einen eigenen Task erstellen?

schreibfaul1, 2022-02-20:
Natürlich, an einem gut funktionierendem Gerät ändert man nichts. In V2 läuft ein separater Audiotask. Daher benötigt der VS1053 einen eigenen SPI (sonst sind die auftretenden Probleme zu groß) und hier eine SD Karte nur für Audiodaten parallel wird sicher funktionieren. 
Ein DAC hingegen ist über I2S angeschlossen und braucht kein SPI.

AVRsteffen, 2022-02-20:
> In V2 läuft ein separater Audiotask. Daher benötigt der VS1053 einen eigenen SPI (sonst sind die auftretenden Probleme zu groß) und hier eine SD Karte nur für Audiodaten parallel wird sicher funktionieren.

Aber man braucht nicht extra einen Task für die Kommunikation mit der SD über den SPI machen, oder? Man kann die SD einfach über SD.h und FS.h auf dem 2. SPI (HSPI) anbinden?

> Ein DAC hingegen ist über I2S angeschlossen und braucht kein SPI.

Ein I2S DAC alá PCM5102A habe ich schon da. Dann wird doch die ganze Decodierung in ESP32 gemacht, oder?

schreibfaul1, 2022-02-23:
Thank you for your interest in V2. You can continue to use the same components, but the pins are different. V2 is still not the master, please proceed as https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/MiniWebRadio-V2/additional_info/How%20to%20install.pdf and change the branch in the lower left corner 
[image]
I wish you success!

AVRsteffen, 2022-04-25:
schreibfaul schrieb:
> Der Wiedereinstieg an der letzten Dateiposition ist etwas schwierig, da es nicht ausreicht den Dateizeiger zu speichern. Es müsste noch eine Korrektur mit den verbleibenen Daten im Audiobuffer erfolgen. Das geht, aber es sind kleine Änderungen in der vs1053ext Lib notwendig. Aber das ist eine gute Idee für die Zukunft. 

> Re-entering at the last file position is a bit tricky, since storing the file pointer is not enough.  A correction would have to be made with the remaining data in the audio buffer.  That works, but small changes in the vs1053ext Lib are necessary.  But that's a good idea for the future.

Wie ist denn hierzu der Stand der Dinge? Hat sich zu diesem Thema da noch was ergeben? Und da fällt mir noch eine Frage an Wolle zum Thema 'mp3' ein. Wie groß (Einträge) kann man die mp3 liste machen? Momentan ist diese auf neun Einträge begrenzt. Also maximal neun mp3 Songs.

What is the current status of this?  Has anything else come up on this topic?  And that reminds me of a question for Wolle on the subject of 'mp3'.  How big (entries) can you make the mp3 list?  This is currently limited to nine entries.  So a maximum of nine mp3 songs.

siffup, 2022-05-09:
I uploaded a couple of MP3 files, but the player stops after the first song. Is it possible to play all MP3 files one after the other?
It also would be great, if I could regulate the volume while playing MP3.
Thank you

closed 2022-08-13 by stale[bot]