schreibfaul1/ESP32-MiniWebRadio #71: Tone control

romulus1st, 2022-02-26:
Hello!
Tone control via the web page doesn't work properly for me, either with VS1053 Featherwing or UDA1334A Breakout. It behaves like the gain and freq sliders are inverted with each other. If I move the gain sliders nothing happens, but if I move the freq sliders the tone changes.

romulus1st, 2022-02-27:
Hello!
I tested the latest version of V2 and the tone control works flawlessly on the windows platform (chrome and edge) but does not work on android platforms (both chrome and edge). I used two android platforms from different manufacturers for testing.

schreibfaul1, 2022-02-27:
Exactly, mousedown mouseup click events are not working in Android. In that case Android Touch Events are the first choice.

romulus1st, 2022-02-28:
Great job, thanks a lot!

closed 2022-02-28 by romulus1st