schreibfaul1/ESP32-MiniWebRadio #72: HILFE über IP

KT315E, 2022-02-27:
Hallo %)
bitte sagt mir jemand:

1. Wie komme ich zur Webseite des Radios?
2. Funktioniert es ohne SD-Karte?

Danke.

schreibfaul1, 2022-02-28:
1) Über den Webbrowser die vom Router zugewiesene IP Adresse aufrufen
2) eine SD Karte ist unbedingt erforderlich, SPIFFS reicht für die Speicherung der Daten nicht aus

KT315E, 2022-03-01:
> 1. Über den Webbrowser die vom Router zugewiesene IP Adresse aufrufen

Und wie Sie es bestimmen (Ich habe nicht überall gesehen, als ob das Panel nicht startet, ESP32-BROOM-32U verwenden)?
Er hat keine eigene Einstiegsadresse, wie bei anderen Projekten (z. B. Karadio)?
Wenn es keine Ierneta gibt, beginnt es um 192.168.1.4, und was ist Ihr Projekt? 

> 2. eine SD Karte ist unbedingt erforderlich, SPIFFS reicht für die Speicherung der Daten nicht aus

SD-Karte ist erforderlich, um das gleiche zu starten?
Ohne es können Sie in die Weboberfläche gelangen?

PS: Vielleicht freue mich die IP-Adresse, die ich aufgrund des Mangels an einer SD-Karte zufrieden bin?!

closed 2022-07-10 by stale[bot]