schreibfaul1/ESP32-MiniWebRadio #73: cs signal sd, V2

ivp1967, 2022-03-03:
Hello.
CS signal SD in the new version is not connected anywhere. I get a message - the SD card is not detected. Maybe the reason is in the CS signal?

schreibfaul1, 2022-03-03:
The SD card is operated in SD_MMC mode. The SD card is connected directly to the ESP32. All resistors against 3.3V or resistors in series must be removed. For example, if you use this display,
[image]
 R1, R2 and R3 are to be shorted.

ivp1967, 2022-03-03:
I solved the problem. It is necessary to pull the GPIO12 to the ground with a 1K resistor. Everything works fine. Many thanks for this wonderful project.

schreibfaul1, 2022-03-04:
GPIO12 is a special case, if pulled high boot fails. I used the default pins for VSPI and HSPI. There is probably a pull-up resistor in your VS1053 board.

closed 2022-08-13 by stale[bot]