schreibfaul1/ESP32-MiniWebRadio #75: Backlight control

romulus1st, 2022-03-05:
Hello!
It would be nice to add backlight control in common.h, to enable or disable the appropriate menu. I added / modified a few lines in the source code for this.

in common.h added line 14
```
#define TFT_BACKLIGHT   1                               // (0) backlight control off, (1) backlight control on
```

in main.cpp, lines 1277 and 1695 changed
```
 if(TFT_BACKLIGHT == 1){
```

Also, the pwm control has a resolution of 8 bits, so it can be between 0 .... 255. With a minimum duty cycle of 5 and a maximum of 100, the backlight control is between 2% - 39%. I propose the use of a duty cycle between 85 - 255, which means a backlight control between 33% - 100%.

```
/***********************************************************************************************************************
*                                        T F T   B R I G H T N E S S                                                   *
***********************************************************************************************************************/
void setTFTbrightness(uint8_t duty){ // duty 85...255 (min...max)
    ledcAttachPin(TFT_BL, 1);        // Configure variable led, TFT_BL pin to channel 1
    ledcSetup(1, 12000, 8);          // 12 KHz PWM and 8 bit resolution
    ledcWrite(1, duty);
}
inline uint32_t getTFTbrightness(){
    return ledcRead(1);
}
inline uint8_t downBrightness(){
    uint8_t br; br = pref.getUShort("brightness");
    if(br > 85) {
        br-=5;
        pref.putUShort("brightness", br);
        setTFTbrightness(br);
        showBrightnessBar();
    } return br;
}
inline uint8_t upBrightness(){
    uint8_t br; br = pref.getUShort("brightness");
    if(br < 255){
        br += 5;
        pref.putUShort("brightness", br);
        setTFTbrightness(br);
        showBrightnessBar();
    }
    return br;
}
```

I also modified the brightness indicator bar to be green at minimum intensity and red at maximum intensity.

```
void showBrightnessBar(){
    uint16_t vol = tft.width() * map(getBrightness(), 85, 255, 0, 100)/100;
    clearVolBar();
    tft.fillRect(_winVolBar.x, _winVolBar.y + 14, vol, 8, TFT_RED);
    tft.fillRect(vol+1, _winVolBar.y + 14, tft.width()-vol+1, 8, TFT_GREEN);
    _f_volBarVisible = true;
}
```

schreibfaul1, 2022-03-06:
Hello @romulus1st,
 thanks for the code and explanations, I don't have any light sensors to develop and test this. This can be a possibility for many users to expand the project.
best regards
Wolle

romulus1st, 2022-03-08:
Hello again!
I'm not talking about light sensors or other sensors. I do not have anything like that. I'm just saying that the values ​​chosen for backlight control should be changed. Also, the backlight control should be able to be activated / deactivated depending on the display type.
Have a nice day!

closed 2022-03-10 by romulus1st