schreibfaul1/ESP32-MiniWebRadio #76: which config is faster? vs1053 or pcm5102

clear-sky, 2022-03-11:
hi 
i had made  edzelf radio that had problems in 320kb/s stations but in your new config seems problem have been solved
but which  config is faster?vs1053 or pcm5102

schreibfaul1, 2022-03-12:
There is nothing wrong with continuing to use an existing VS1053 board. There are audio boards from various manufacturers complete with DAC and amplifiers that, with a little skill, are good for building internet radios. Hum interference or background noise are then no longer an issue. I don't see a speed advantage. Both variants have advantages and disadvantages

clear-sky, 2022-03-12:
hi
thanks for the answer!
 will be thankful if tell me  advantages and disadvantages

closed 2022-08-13 by stale[bot]