schreibfaul1/ESP32-MiniWebRadio #77: Waveshare 3.5inch RPi LCD (B) Rev2.0

romulus1st, 2022-03-12:
Hello!
It would be nice to add support for Waveshare 3.5inch RPi LCD (B) Rev2.0. This IPS display uses the ILI9486L controller, has a backlight control pin, but requires different init code tables for positive and negative GAMMA control.

```
        writeCommand(0xE0);
        SPI.write16(0x00);
        SPI.write16(0x04);
        SPI.write16(0x0E);
        SPI.write16(0x08);
        SPI.write16(0x17);
        SPI.write16(0x0A);
        SPI.write16(0x40);
        SPI.write16(0x79);
        SPI.write16(0x4D);
        SPI.write16(0x07);
        SPI.write16(0x0E);
        SPI.write16(0x0A);
        SPI.write16(0x1A);
        SPI.write16(0x1D);
        SPI.write16(0x0F);

	writeCommand(0xE1);
        SPI.write16(0x00);
        SPI.write16(0x1B);
        SPI.write16(0x1F);
        SPI.write16(0x02);
        SPI.write16(0x10);
        SPI.write16(0x05);
        SPI.write16(0x32);
        SPI.write16(0x34);
        SPI.write16(0x43);
        SPI.write16(0x02);
        SPI.write16(0x0A);
        SPI.write16(0x09);
        SPI.write16(0x33);
        SPI.write16(0x37);
        SPI.write16(0x0F);
```

schreibfaul1, 2022-03-13:
hello @romulus1st,
thanks for the hint. I don't think this is just a problem with Waveshare, it also depends on the installed display.
[image]
I added 'DISPAY_INVERSION' for all TFT controllers, hope it works

romulus1st, 2022-03-13:
Hello ! Thanks for the reply. Yes, this display requires DISPLAY INVERSION to be ON. I already did this on tft.cpp.
```
//        writeCommand(0x20); // Display Inversion OFF   RPi LCD (A)
          writeCommand(0x21); // Display Inversion ON    RPi LCD (B)
```
But the colors are not OK.
The first image is with INVON = '1' and the default GAMMA tables.

[image]

The second image is with INVON = '1' and the GAMMA tables used by the recommended Raspbian driver for this display.

[image]

Thanks for the best web radio project I've tried so far!

schreibfaul1, 2022-03-13:
Too bad that's not the expected result. What if your gamma curve is considered to be inverse and it doesn't have the DisplayInversion bit set (only for ILI9486). What do you make of it?. I don't want to build in a separate gamma curve for every display on the market.

romulus1st, 2022-03-13:
OK, I understand. I thought it would be useful for others who may have the same problem.

closed 2022-03-13 by romulus1st

schreibfaul1, 2022-03-13:
I think you misunderstood, your gamma correction is built in. Hardly anything has changed on my display (RPI display but no Waveshare). The green is a little clearer. Can you please see how it looks on your display?

romulus1st, 2022-03-13:
With built in gamma correction, blue is slightly stronger. Probably in the case of ILI9486, which is manufactured in both TN and IPS technologies, the GAMMA correction should be left  to default or 'built in' to have a library compatible with any display ?