schreibfaul1/ESP32-MiniWebRadio #78: Repository for MiniWebRadio-V1

OK1FWG, 2022-03-29:
Hi,
whatever I try, I am no longer able to open/clone the V1 repository. Everytime I get the V2 source. Any clue?

BTW, I've build the V2, uploaded the code and it is running really very good. Just having issues with the touchpad - I have the ILI9341 display with the HR2046 (or similar) controller. Regardless of any setting I used, the touch regions are somewhere else than on the buttons. Also, when on the Search page, I can choose Country etc., radio start playing stream after 2 or more clicks on the StreamURL button, Stream Title change, but not the logo (in the web interface it is okay). Is this fully implemented or it's just not there yet?
Anyway, great job, keep coding :-)

schreibfaul1, 2022-03-29:
Too bad that you have such problems with the V2. I myself have an ILI9341 display which works very well. Have you tried other values ​​for TP_VERSION or TP_ROTATION in common.h? Displays with 320x480px can now also be used. There are larger fonts and logos for this. To do this I had to adjust the directory tree on the SD. Please also download 'Content_on_SD_Card.zip' and extract it to the SD card.
Unfortunately I didn't assign a tag for the V1. You can find old files in the history before February.

OK1FWG, 2022-03-29:
That's all okay, the SD card is working good and really fast, that's fine. I know about the possibility to use a larger display, as you mentioned. I did tried several possible combination of those settings (TP_VERSION & TP_ROTATION), but no avail. The same display worked well in the V1. I'll give it another try.
Thanks about the info for V1 files.

nodoubtman, 2022-06-22:
> Hi, whatever I try, I am no longer able to open/clone the V1 repository. Everytime I get the V2 source. Any clue?
>
> BTW, I've build the V2, uploaded the code and it is running really very good. Just [...]

I have just uploaded it for you:

https://file.io/2c2er5UAG5lT

Thank you.
Marc.

schreibfaul1, 2022-06-22:
Hello Marc,
Unfortunately, I did not create a tag for version 1, so only browsing in the history remains.
What display do you have? Does setting the values ​​in common.h (TP_VERSION, TP_ROTATION) fail?
The search page brings up the results of the Community Radio Browser. And the button to the right of the StreamURL is only for testing the link (works?, link is broken or you don't like it). A logo is often included. You can also add your own e.g. from Google image search via URL. The name of the logo must match the name of the station, otherwise the logo will not be found afterwards. Then you load the picture onto the SD card (arrow up) and the station information with the arrow left into the station list. If that's okay, save the changed channel list to the SD card with the save button. The list can contain a maximum of 999 stations. You can also save a backup as an Excel list on your PC, you never know. You can also edit the list in Excel or Libre Office and write it back to the SD card. The SD card never needs to be removed from the adapter. You can upload files (e.g. mp3 files) with Filezilla or another FTP program.

best regards
Wolle

nodoubtman, 2022-08-25:
ok

clear-sky, 2023-04-14:
https://github.com/dorinik/ESP32-MiniWebRadio

closed 2024-03-12 by github-actions[bot]