schreibfaul1/ESP32-MiniWebRadio #79: Which partition scheme should be used for a 16MB ESP32 board?

dreamy1, 2022-04-03:
Hello,

thanks for this great work!

I am confused how to start with the arduino IDE with several partition schemes to choose, for a ESP32 devkit board there are following options e.g.:

[image]

Which option is the right one to choose?

Thanks,
Stefan

schreibfaul1, 2022-04-03:
Hello @dreamy1,
we need a special partition table with large nvs. The data is in miniwebradio.cvs. PlatformIO takes care of the correct partitioning for us. Another IDE requires some manual configuration.

best regards
Wolle

dreamy1, 2022-04-03:
Hello Wolle,

thank you very much - so it will create a manual partition table in the IDE, this should work.

Thanks for quick response!
Stefan

dreamy1, 2022-04-03:
Ich sehe gerade dass wir wohl beide aus D kommen :-)

Darf ich noch fragen, ob ein LAN bzw. Ethernet Support geplant ist? Ich habe hier gerade eine Version mit LAN8720 designt.

Viele Grüße,
Stefan

schreibfaul1, 2022-04-03:
Um nochmal auf die Partition zurückzukommen. Du kannst die csv in /tools/partitions kopieren und die boards.txt für dein Board erweitern. Dann ist die neue Partition in der IDE auswählbar.
Mil LAN habe ich bisher nichts gemacht, wenn es dann bei dir funktioniert kannst Du gerne einen PR einstellen.
vG
Wolle

dreamy1, 2022-04-03:
Hallo Wolle,

vielen Dank für die Rückmeldung - so hatte ich das vor, das ist sicher der beste Weg mit der Erweiterung der boards.txt.

Ich versuche das mit LAN mal ans Laufen zu bekommen.

Lieben Dank für Deine Hilfe und viele Grüße,
Stefan

nodoubtman, 2022-06-06:
> Hello @dreamy1, we need a special partition table with large nvs. The data is in miniwebradio.cvs. PlatformIO takes care of the correct partitioning for us. Another IDE requires some manual configuration.
> 
> best regards Wolle

Hello.
Can you explain in details how to do it with Arduino IDE.

Thank you.
Marc.

schreibfaul1, 2022-06-06:
Hello Marc,
the problem is actually the partition table. I haven't worked with the Arduino IDE for a long time. The easiest way might be to rewrite an existing partition table. Take the content of the miniwebradio.csv and copy it into the huge_app.csv. A simple editor is sufficient for this. For this it is necessary to find out where the Arduion IDE saves the current framework.

The easy way: Install VSCode. VSCode is available for many operating systems. Then add the PlatformIO extension. There is not much more to do. PlatformIO does everything else for you.

Best regards
Wolle

closed 2023-04-14 by stale[bot]