schreibfaul1/ESP32-MiniWebRadio #80: WiFi credentials are not correct

valera39, 2022-04-04:
Hello!
Finally I waited for the update and decided to repeat your radio.
I can’t connect to the network, the common.h file was corrected, I also added my network to networks.csv. Tried several esp 32 which I removed from working radios. Here is the log:
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
00:00:00 [0m
00:00:00 [33m***************************[0m
00:00:00 [33m*     MiniWebRadio V2     *[0m
00:00:00 [33m***************************[0m
00:00:00 [0m
00:00:00 tft_info: init [36mILI9341[0m
00:00:00 setup: Init SD card[0m
00:00:00 [37msetup: SD card found[0m
00:00:00 setup: seek for stations.csv[0m
00:00:00 setup: stations.csv found[0m
00:00:00 setup: seek for WiFi networks[0m
00:00:00 WiFI_info: Connecting WiFi...[0m
[0;31m[ 10859][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).[0m
00:00:10 [31mWiFi credentials are not correct
[0m
00:00:10 [31mWiFi credentials are not correct[0m
What else can be done? Your previous project worked immediately and works fine to this day
Thanks,

schreibfaul1, 2022-04-04:
In V1 SSID/PW was a string, now it's a #define directive. Maybe there are problems with special characters. Please enter the credentials in networks.csv, maybe it will help. You can set the value in -DCORE_DEBUG_LEVEL to 5 in platformio.ini, for a debug log.

valera39, 2022-04-04:
Thanks for the quick response! I prescribed
#define _SSID "Honor" // Your WiFi credentials here
#define _PW "12345678" and in networks.csv as written in the instructions, but no result

valera39, 2022-04-04:
here is what i got
`COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
[0;37m[     0][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=21, ...)[0m
[0;37m[     0][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=50009, ...)[0m
[0;37m[     0][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)[0m
[0;37m[     0][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServe       ����80, ...)[0m
[0;36m[     1][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz[0m
00:00:00 [0m
00:00:00 [33m***************************[0m
00:00:00 [33m*     MiniWebRadio V2     *[0m
00:00:00 [33m***************************[0m
00:00:00 [0m
00:00:00 tft_info: init [36mILI9341[0m
00:00:00 setup: Init SD card[0m
00:00:00 [37msetup: SD card found[0m
00:00:00 setup: seek for stations.csv[0m
00:00:00 setup: stations.csv found[0m
00:00:00 setup: seek for WiFi networks[0m
[0;32m[   811][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U[0m
[0;32m[   821][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U[0m
[0;32m[   824][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_01[0m
[0;32m[   831][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_02[0m
[0;32m[   839][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_03[0m
[0;32m[   847][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04[0m
[0;32m[   854][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: network_04[0m
00:00:00 WiFI_info: Connecting WiFi...[0m
[0;36m[   886][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY[0m
[0;37m[   974][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started[0m
[0;36m[   976][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START[0m
[0;37m[  3607][V][WiFiGeneric.cpp:306] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 7[0m
[0;36m[  3608][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 1 - SCAN_DONE[0m
[0;32m[  3614][I][WiFiMulti.cpp:114] run(): [WIFI] scan done[0m
[0;32m[  3620][I][WiFiMulti.cpp:119] run(): [WIFI] 7 networks found[0m
[0;36m[  3626][D][WiFiMulti.cpp:149] run():  --->   0: [7][2C:FD:A1:C2:44:F0] ASUS_AC66U (-59) *[0m
[0;36m[  3635][D][WiFiMulti.cpp:151] run():        1: [4][30:B5:C2:E3:45:E0] TP-LINK_E345E0 (-71) *[0m
[0;36m[  3644][D][WiFiMulti.cpp:151] run():        2: [5][D8:07:B6:B7:FE:0F] GG (-78) *[0m
[0;36m[  3652][D][WiFiMulti.cpp:151] run():        3: [3][2C:FD:A1:03:D7:60] ASUS_60 (-83) *[0m
[0;36m[  3661][D][WiFiMulti.cpp:151] run():        4: [5][10:FE:ED:DC:7C:A8] TP-LINK_DC7CA8 (-85) *[0m
[0;36m[  3670][D][WiFiMulti.cpp:151] run():        5: [13][A8:63:7D:13:92:6E] Eva22 (-88) *[0m
[0;36m[  3679][D][WiFiMulti.cpp:151] run():        6: [3][14:D6:4D:37:DD:B6] Family (-92) *[0m
[0;32m[  3687][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 7 (-59)[0m
[0;37m[  3700][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0[0m
[0;31m[  8728][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).[0m
00:00:08 [31mWiFi credentials are not correct
[0m
00:00:08 [31mWiFi credentials are not correct[0m
[0;37m[  9395][V][WiFiGeneric.cpp:289] _arduino_event_cb(): STA Disconnected: SSID: ASUS_AC66U, BSSID: 2c:fd:a1:c2:44:f0, Reason: 15[0m
[0;36m[  9396][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED[0m
[0;33m[  9405][W][WiFiGeneric.cpp:852] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT[0m
`

valera39, 2022-04-04:
It turned out to connect, but there was another problem: the touchscreen does not work. I tried two displays ILI9341 and ILI9488, both have a touchscreen on the XPT2046. Tried various variations of #define TP_VERSION but no luck. In the previous radio version, there is a similar ILI9341. And where are the display pins T_D0, T_DIN and T_CLK connected?
Thank you, I'm looking forward to your reply.

schreibfaul1, 2022-04-05:
this is a known problem with the ILI9488 display, pin SDO(MISO) must remain free
https://github.com/Bodmer/TFT_eSPI/issues/705
[image]

valera39, 2022-04-05:
Hello!
Thanks for making time for me
ILI9488 display, connected as in your diagram, SDO (MISO) and T_D0 did not connect to GPIO 19 ESP32,
#define TP_VERSION 4 // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486RPI, (4)ILI9488, but still the touchscreen doesn't work.
The display is working, checked. Like this: https://aliexpress.ru/item/1005001999296476.html?item_id=1005001999296476&sku_id=12000018365356571

schreibfaul1, 2022-04-05:
Hi,
i use this one (ILI9488 Display): https://www.amazon.de/gp/product/B08GBYQQV5/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1
and on this TFT_MISO is not tristate.

[image]

I can't say anything about the display from AliExpress, maybe someone knows more.

closed 2022-04-05 by valera39