schreibfaul1/ESP32-MiniWebRadio #81: main.cpp and other questions

valera39, 2022-04-05:
Hello!
Thanks to your tips and selection of options in the TFT_ROTATION, TP_VERSION, TP_ROTATION settings, I managed to launch my display.
If you don't mind, I have a few more questions.
- In the main.cpp file with the PlatformIO editor, I try to change lines 75-84 to
"** интернет-радио **",         // "* интернет-радио *"  "ραδιόφωνο Internet"
                "** интернет-радио **",
                "** интернет-радио **",
                "** часы́ **",                    // Clock "** часы́ **"  "** ρολόι **"
                "** часы́ **",
                "** яркость **",             // Brightness яркость λάμψη
                "** цифрово́й плеер **",            // "** цифрово́й плеер **"
                "** цифрово́й плеер **",
                "" ,                            // Alarm should be empty
                "* Таймер сна *",       // "Sleeptimer" "Χρονομετρητής" "Таймер сна"
but when compiling in terminal the error is:
Compiling .pio\build\esp32dev\src\main.cpp.o
src/main.cpp:85:1: error: initializer-string for array of chars is too long [-fpermissive]
 };
 ^
src/main.cpp:85:1: error: initializer-string for array of chars is too long [-fpermissive]
src/main.cpp:85:1: error: initializer-string for array of chars is too long [-fpermissive]
src/main.cpp:85:1: error: initializer-string for array of chars is too long [-fpermissive]
src/main.cpp:85:1: error: initializer-string for array of chars is too long [-fpermissive]
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
what am I doing wrong?
- also in lines 238 I change to pref.putUShort("volume",5); // 0...21
and at 241 to pref. putUShort("brightness", 50); // 0...100
but when I turn it back on, my settings don't work. In the previous version, I edited and compiled without problems.
Or is it because of PlatformIO?
- how to change the settings of the remote control buttons in this version using your own remote control?
- Can IR VS1838B be used?
- is it possible to turn off voice messages every hour?
Sorry for many questions.
Thanks!

schreibfaul1, 2022-04-05:
Compiler error 'error: initializer-string for array of chars is too long' is eliminated. Cyrillic letters take up more memory than Latin ones.
'volume' and 'brightness' in defaultsettings are only called up the first time, after that the variables are known in the NVS and are permanently saved. After a restart, the last values ​​are restored.
IR is this: https://github.com/schreibfaul1/ESP32-IR-Remote-Control/blob/master/images/RemoteControl.jpg
For a different remote control, the values ​​in the table must be changed.
VS1838B can be used
I will turn off the announcement of the time with the next update.

romulus1st, 2022-04-08:
Hi, I have modified IR.cpp for this type of remote control.

[image]

If you use this type of remote control, the changes are:
- line 112:
```
if((intval >= 3500)&&(intval <= 5500)) {    // begin sequence of code?
```
        
- line 131 - 147:
```
                case 39015: ir_resp=0;  break; //ZERO   0x9867
                case 41565: ir_resp=1;  break; //ONE    0xA25D
                case 25245: ir_resp=2;  break; //TWO    0x629D  
                case 57885: ir_resp=3;  break; //THREE  0xE21D
                case 8925:  ir_resp=4;  break; //FOUR   0x22DD
                case 765:   ir_resp=5;  break; //FIVE   0x02FD
                case 49725: ir_resp=6;  break; //SIX    0xC23D
                case 57375: ir_resp=7;  break; //SEVEN  0xE01F
                case 43095: ir_resp=8;  break; //EIGHT  0xA857
                case 36975: ir_resp=9;  break; //NINE   0x906F
                case 14535: ir_resp=10; break; //OK     0x38C7
                case 6375:  ir_resp=11; break; //UP     0x18E7
                case 19125: ir_resp=12; break; //DOWN   0x4AB5
                case 23205: ir_resp=13; break; //RIGHT  0x5AA5
                case 4335:  ir_resp=14; break; //LEFT   0x10EF
                case 45135: ir_resp=15; break; //HASH   0xB04F
                case 26775: ir_resp=16; break; //STAR   0x6897
```

- line 154:
```
    if((intval > 400) && (intval < 750)){       // Short pulse?
```

- line 158:
```
    if((intval > 1400) && (intval < 1900)){     // Long pulse?
```

schreibfaul1, 2022-04-08:
Three different IR transmitters, three different button assignments. I incorporated that and increased the time intervals of the impulses :-)

closed 2022-05-08 by valera39